knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #253: [BUG] (Kann mit RGB bzw RGBW Funktion nicht kompiliert werden)

corradodave, 2022-01-05:
Wenn
`//#define RGB_LED`
oder
`//#define RGBW_LED`

Unkommentiert wird, kann nicht kompiliert werden.

```
<artificial>:(.text+0x5758): undefined reference to `PrinterEventLEDs::onHotendHeating(int, int, int)'
<artificial>:(.text+0x5a4e): undefined reference to `LEDLights::set_color(LEDColor const&)'
<artificial>:(.text+0x5ac2): undefined reference to `PrinterEventLEDs::old_intensity'
<artificial>:(.text+0x5cec): undefined reference to `PrinterEventLEDs::onBedHeating(int, int, int)'
<artificial>:(.text+0x5e4a): undefined reference to `PrinterEventLEDs::old_intensity'
C:\Users\corra\AppData\Local\Temp\ccDhowL9.ltrans0.ltrans.o: In function `GcodeSuite::process_parsed_command(bool)':
<artificial>:(.text+0xd568): undefined reference to `GcodeSuite::M150()'
<artificial>:(.text+0xdac0): undefined reference to `LEDLights::set_color(LEDColor const&)'
<artificial>:(.text+0xdac8): undefined reference to `PrinterEventLEDs::leds_off_after_print'
<artificial>:(.text+0xdaec): undefined reference to `PrinterEventLEDs::leds_off_after_print'
<artificial>:(.text+0xdafe): undefined reference to `LEDLights::set_color(LEDColor const&)'
<artificial>:(.text+0xdb04): undefined reference to `PrinterEventLEDs::leds_off_after_print'
C:\Users\corra\AppData\Local\Temp\ccDhowL9.ltrans1.ltrans.o: In function `LEDLights::get_color()':
<artificial>:(.text+0x35bc): undefined reference to `LEDLights::lights_on'
<artificial>:(.text+0x35c4): undefined reference to `LEDLights::color'
<artificial>:(.text+0x35c8): undefined reference to `LEDLights::color'
<artificial>:(.text+0x35cc): undefined reference to `LEDLights::color'
<artificial>:(.text+0x35d0): undefined reference to `LEDLights::color'
C:\Users\corra\AppData\Local\Temp\ccDhowL9.ltrans1.ltrans.o: In function `Temperature::wait_for_bed(bool)':
<artificial>:(.text+0x3630): undefined reference to `PrinterEventLEDs::old_intensity'
<artificial>:(.text+0x37b0): undefined reference to `PrinterEventLEDs::onBedHeating(int, int, int)'
C:\Users\corra\AppData\Local\Temp\ccDhowL9.ltrans1.ltrans.o: In function `Temperature::wait_for_hotend(unsigned char, bool)':
<artificial>:(.text+0x3996): undefined reference to `PrinterEventLEDs::old_intensity'
<artificial>:(.text+0x3b0e): undefined reference to `PrinterEventLEDs::onHotendHeating(int, int, int)'
<artificial>:(.text+0x3c40): undefined reference to `LEDLights::set_color(LEDColor const&)'
C:\Users\corra\AppData\Local\Temp\ccDhowL9.ltrans1.ltrans.o: In function `main':
<artificial>:(.text.startup+0x418): undefined reference to `LEDLights::setup()'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\MEGA_S_TMC_BLT_10\firmware.elf] Error 1
================================================================================================== [FAILED] Took 21.83 seconds ==================================================================================================

Environment        Status    Duration
-----------------  --------  ------------
MEGA_S_TMC_BLT_10  FAILED    00:00:21.830
============================================================================================= 1 failed, 0 succeeded in 00:00:21.830 =============================================================================================

```

corradodave, 2022-01-06:
Besteht evtl eine andere Möglichkeit die PINs der Servoports (4,5,6,12) anzusteuern?

stklcode, 2022-01-15:
Case light über die servo pin 2 ist sogar default, muss man nur aktivieren. So habe ich meine Beleuchtung angeschlossen, steuerbar über [M355](https://marlinfw.org/docs/gcode/M355.html)

https://github.com/stklcode/Marlin-2-0-x-Anycubic-i3-MEGA-S/commit/ce6fb5330b8571507a5e4f97f5e95ccdb88b4325

Zum eigentlichen Problem:
Die benötigten Bibliotheken sind beim Build ausgeschlossen. Dazu müsste man einmal die _platformio.ini_ durchsehen und die Excludes nach Bedarf auskommentieren:
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/7b9e0c4e023b4533ff435551ded8a4041e2a64ba/platformio.ini#L173-L179

corradodave, 2022-01-15:
Hi, Danke für die Antwort, 
ja ist nur ein RGBW Streifen, soll beim Aufheizen Rot, Beim Druck Weiß und nach dem Druck Grün Leuchten.

Werde es mal in der Platfdormio.ini probieren danke

stklcode, 2022-01-15:
Jetzt werden die Module ausgeschlossen, darum steht ein `-` davor. Wenn du die Zeile auskommentieret wird es nicht mehr ausgeschlossen, also mit gebaut ;)

corradodave, 2022-01-15:
ne geht leider nicht, da kommen dann tausend andere Fehler :(

stklcode, 2022-01-16:
In den Fehlremeldungen steht doch recht detailliert, was fehlt ... GCode M150

Damit kompiliert es:

**configuration_adv.h**

```c
#define RGB_LED
#define RGB_LED_R_PIN 6
#define RGB_LED_G_PIN 5
#define RGB_LED_B_PIN 12
```

**platformio.ini.h**

```
#  -<src/feature/leds/leds.cpp>
...
#  -<src/gcode/feature/leds/M150.cpp>
```

corradodave, 2022-01-16:
Geht nicht, Kollidiert dann immer mit dem Filament Sensor
```

Compiling .pio\build\MEGA_S_TMC_BLT_10\src\src\feature\bltouch.cpp.o
Compiling .pio\build\MEGA_S_TMC_BLT_10\src\src\feature\cancel_object.cpp.o
Compiling .pio\build\MEGA_S_TMC_BLT_10\src\src\feature\caselight.cpp.o
Compiling .pio\build\MEGA_S_TMC_BLT_10\src\src\feature\controllerfan.cpp.o
Compiling .pio\build\MEGA_S_TMC_BLT_10\src\src\feature\e_parser.cpp.o
Compiling .pio\build\MEGA_S_TMC_BLT_10\src\src\feature\host_actions.cpp.o
Marlin\src\MarlinCore.cpp: In function 'void event_filament_runout()':
Marlin\src\MarlinCore.cpp:424:63: error: cannot convert 'const char*' to 'PromptReason' for argument '1' to 'void host_action_prompt_begin(PromptReason, const char*, char)'
       host_action_prompt_begin(PSTR("FilamentRunout T"), false);
                                                               ^
Compiling .pio\build\MEGA_S_TMC_BLT_10\src\src\feature\leds\leds.cpp.o
Compiling .pio\build\MEGA_S_TMC_BLT_10\src\src\feature\leds\printer_event_leds.cpp.o
*** [.pio\build\MEGA_S_TMC_BLT_10\src\src\MarlinCore.cpp.o] Error 1
========================================================================================== [FAILED] Took 4.69 seconds ==========================================================================================

Environment        Status    Duration
-----------------  --------  ------------
MEGA_S_TMC_BLT_10  FAILED    00:00:04.690
==================================================================================== 1 failed, 0 succeeded in 00:00:04.690 ==================================================================================== 
Der Terminalprozess "C:\Users\corra\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'MEGA_S_TMC_BLT_10'" wurde mit folgendem Exitcode beendet: 1.

Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste.
```

stklcode, 2022-01-16:
Benötigst du "printer_event_leds"?

corradodave, 2022-01-16:
Also hab jetzt mal bisschen Rumprobiert, das geht alles soweit mit den LED`s

Hab mir mal ne Frische Knutwurst version geladen und sobald man die Zeile 2014 
`//#define FILAMENT_RUNOUT_SENSOR`

Einkommentiert, gibts da nen Fehler, hat also garnichts mit den LEDS zutun.

knutwurst, 2022-01-24:
@corradodave der Filamentsensor wird in dieser Firmware auch über die Touchscreen-Routine gesteuert. Die Zeile von dir darf somit nicht einkommentiert werden.

corradodave, 2022-01-24:
Dann passt es, LED Feature geht wunderbar mit:

configuration.h

#define RGBW_LED

platformio.ini
#  -<src/feature/leds/leds.cpp>
#  -<src/gcode/feature/leds/M150.cpp>

closed 2022-01-24 by corradodave