knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #254: [FR] Pre-builds for alternative, easier BLTouch installation (Pin D21)

sapphyrus, 2022-01-09:
### Description

I recently upgraded my Anycubic i3 Mega S with a BLTouch, specifically [this kit](https://www.amazon.de/gp/product/B08M63NWJP) (which was 48€ at the time of purchase). It already came with extensions that allowed me to just route both, the 3-pin and 2-pin to the mainboard and connect them without soldering anything.
For that I followed [this guide](https://youtu.be/b8yWIWXus1U) by OsLee540i that uses the SCL/D21 and T3/A12 pins for the 2-pin connector. These were easy to connect with some jumper wires. The 3-pin servo connector was also already wired up correctly.

All in all this allowed for a really easy installation of the BLTouch with no soldering at all, which is why I'm opening this issue. Currently this requires building the firmware yourself and setting `Z_MIN_PROBE_PIN` to 21. That's not a big issue for people familiar with programming, but can't really be expected from a normal end user. That is why I think there should be pre-builds for this configuration or some way to pick the BLTouch pin dynamically, like setting it via gcode or trying both pins during auto leveling. Unfortunately I'm not very familiar with Marlin so I have no idea if what I'm suggesting is even possible.

If that's done the wiki page should also be updated to include this alternative way since it really lowers the barrier of entry for this upgrade.

corradodave, 2022-01-09:
Hey, you dont need the T3/A12 Pin, its Only a Ground and not needed.
You can use the only white wire and connect do any Input Pin.

A Preebuild for all Pins is crazy, just config the Pin and compile it

sapphyrus, 2022-01-09:
> A Preebuild for all Pins is crazy, just config the Pin and compile it

Obviously prebuilds for all pins would be crazy, just saying that this configuration is IMO better than the one currently listed in the wiki.

SpRiNgToIfEl, 2022-01-23:
I also use the PIN21 for the BLTouch.
Very simple implementation and no soldering necessary.
Easy implementation of a BLTouch on a Mega (X).
**### Thanks for the tip from OSLee540i ###**

Activate the following lines in configuration.h:
#define USE_PROBE_FOR_Z_HOMING
#define Z_MIN_PROBE_PIN 21

Disable the following line in configuration.h:
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

_Note: When compiling I had the following error message:_
"Z_MULTI_ENDSTOPS is incompatible with USE_PROBE_FOR_Z_HOMING".

Disable the following line in configuration_adv.h:
//#define Z_MULTI_ENDSTOPS

After that everything works as it should.

I think the above steps should be activated / deactivated automatically when you select the BLTouch... [BUG?]

knutwurst, 2022-01-23:
This is not a bug, but intentional. Many users use PIN 2 as a trigger pin, as this is already present in the wiring harness. I like the soldering better than fiddling with an additional cable.

One more note:
If you use the Z-Probe as an endstop, you must use G34 (stepper auto align) otherwise the bed may hang crooked.

You got around this by disabling the second endstop.
This removes the printer's ability to level the bed and I'll bet it'll get more and more crooked over time.

You can do it like this, but I won't include it in the official precompiled version.

SpRiNgToIfEl, 2022-01-23:
yes you are right, g34 must be used, see this [posting ](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/266)>>, i forgot to mention that.
a note in the FW (or your FAQ) would be helpful for some users, if you want to disable the mechanical z-endstops.

for all those who are interested:
G34 = Z_STEPPER_AUTO_ALIGN (incl. z-home after autoalign!)

why is G34 not used as standard for axis alignment when BLTouch is activated?
if you don't trust BLTouch to align the z/x-axis, you can't trust it to level the nozzle/bed...

I personally prefer to use the BLTouch completely, it works now without errors.
in my case, the z-axis likes to move, either by gravity or by changing the filament or similar.
(the z-axes are very smooth-running >>> i have often been annoyed because the Z-axis was "magically" misaligned overnight)

Regarding the additional cable:
if you convert the mega x to BMG, as in my case, you can use drag chains for cable routing.
it doesn't matter if there is 1 cable more or less. you can eliminates the z-endstops on both z-axis.

stklcode, 2022-01-23:
Bed leveling and axis alignment are quite different things. There are reasons to handle both separately, i.e. determine axis position within the machine frame with endstops first (the BLTouch obviously can’t handle this, at least not if mounted in the documented position) and measure bed alignment relative to this “zero“ with the probe afterwards.
I wouldn’t prefer a spring-loaded bed as ground truth over a switch bolted to an aluminum frame. Personally never had any issues with misalignment that could not be handled with default mechanisms. Don’t know if that is just luck. Of course if one or another lack of reliability, you have to find an alternative way.

Long story short, I don’t think one option us universally “better“ than the other.

closed 2022-03-01 by github-actions[bot]