knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #255: [FR] Disable Startup Sound

mjaschen, 2022-01-12:
### Preamble

I'd like to thank you for your awesome work on this firmware and the excellent documentation.

### Description

When turning the printer on, a startup sound plays. It's a lot less annoying than the original Anycubic sound, but still … I'd suggest to add an option to disable that sound.

### Feature Workflow

1. Turn printer on
2. No startup sound plays.
3. Zen.

stklcode, 2022-01-14:
If you build firmware yourself, it’s as simple as commenting out this line:
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/7b9e0c4e023b4533ff435551ded8a4041e2a64ba/Marlin/Configuration.h#L2808

Secret menu option to en-/disable it and save the state in the EEPROM might be implemented for user control. Should not be much more effort than maintaining 116 pre-compiled variants (currently 58, all with and without chime)

mjaschen, 2022-01-15:
Hi Stefan,

thanks, that's a solution which works good enough for me!

closed 2022-01-15 by mjaschen