knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #258: Frage für anycubic i3 mega S beheizten Bauchraum

Pkahl, 2022-01-15:
Wäre es möglich bei Marlin einzustellen dass der zweite Druckdüsen Anschluss auf den Mainboard für den beheizten Bauraum benutzen kann mit MOSFET natürlich

knutwurst, 2022-01-24:
Ja.

Pkahl, 2022-01-24:
Und wo muss ich das bei Marlin einstellen

knutwurst, 2022-01-24:
Naja, du musst die Heizung einschalten:
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/66eeb0d7ec151c3e1615fb656140fab5986c08af/Marlin/Configuration.h#L808

Und dann noch den Pin des zweiten Hotends setzen, indem du in die Datei pins_RAMPS.h die Zeile einfügst:
`TEMP_CHAMBER_PIN RAMPS_D9_PIN`

Wie genau dein Bauraum beheizt ist, weiß ich ja nicht. Das musst du dann selbst konfigurieren.

Siehe auch: https://3dprinting.stackexchange.com/questions/10907/how-to-set-up-marlin-to-use-a-heated-chamber

closed 2022-02-24 by knutwurst