knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #259: Mega Pro Laser 1.4 beta Maximum Intesity

Nickel-05, 2022-01-16:
### Bug Description

Started experimenting with the new laser feature and setting the maximum intensity of 255 turns laser off.  

### Steps to Reproduce

From the Terminal, use M3 S0-S255 to set the laser from min to max intensity.  Works from 0-254 but S255 turns the laser OFF instead of Maximum Intensity.

#### Additional Information

stklcode, 2022-01-17:
Same here. No idea yet, why this happens. I simply configured 254 as maximum in the machine settings of my engraving tool as a workaround.

oooBuGooo, 2022-02-19:
> Hier gilt das gleiche. Noch keine Ahnung, warum das passiert. Als Workaround habe ich in den Maschineneinstellungen meines Gravierwerkzeugs einfach 254 als Maximum konfiguriert.

Das habe ich leider auch schon probiert.
Das Problem ist es das in der unteren Werten S1 - S100 keinen großen Unterschied gibt.
Erst in den oberen  S Werten steigt die Intensität des Lasers.

stklcode, 2022-02-20:
Was hat das eine mit dem anderen zu tun?

Laut Oszilloskop produzieren S0-S254 linear ansteigende PWM Tastraten zwischen 0 und 100% am Ausgang. Leichte Verzerrung im Betrieb, dass der Laser-Lüfter standardmäßig parallel zur Laserdiode angeschlossen ist. Aber dafür kann die Software nichts.
Den Unterschied zwischen 25, 50, 75, 100 sehe ich mit bloßem Auge durch die Schutzbrille. Ja, darüber wird es offensichtlich heller, das würde ich nach der Messung aber auf das Auge schieben und nicht auf den Laser.

Gerade einmal in ein paar G-code Daten nachgesehen, was _LightBurn_ so generiert hat. Da habe ich Werte zwischen 40 und 200, was bei Karton mit 50mm/s etwa die Spanne hellgrau bis Schnitt abbildet.

closed 2022-03-28 by github-actions[bot]