knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #262: [BUG] Chiron: Resume after Filament Runout not possible 1.4.0.b4

Starbreezer666, 2022-01-20:
### Bug Description

After replacing the filament there is no possibility to resume the print. Acknowledging the filament runout dialog and pressing Continue afterwards does not continue the print. Pressing Pause afterwards seems to stop the print completely.

### Steps to Reproduce

1. Start print
2. Let Filament run out
3. Replace Filament (No dialog or special menu option used for this)
4. Acknowledge message box
5. (Optional): Press Continue
6. (Optional): Press Pause

**Expected behavior:** After pressing Continue print should resume.

**Actual behavior:** Print head remains stationary. Print does not resume.

#### Additional Information

- Probable same as #250 but this title is more precise.
- Anycubic Display 1.0 is used
- TMC drivers and according firmware is used

Starbreezer666, 2022-01-21:
If there is anything I can do to gather more debug data or help otherwise, please don't hesitate to tell me so :-)

knutwurst, 2022-01-24:
It should be fixed in 1.4.0 Beta 5 ;) 

I had a lot to change to make pause/reheat/resume work, but it should for all printers.

Please open an new ticket, if this (or another) issue persists.

closed 2022-01-24 by knutwurst