knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #263: [FR] Bitte den Thermistor korrigieren!

alfrankgit, 2022-01-21:
Wie schon mehrfach von mir und Anderen erläutert und mit Bildern belegt:

Der ab Werk eingetragene Thermistor Typ 5 ist falsch!

Verbaut ist Typ 11.

In allen Anycubic FDMs. mit DWIN (oder Klon) Display, die mir bisher persönlich untergekommen sind, das wären fünf 4Max Pro, ein 4Max Pro 2.0, ein Mega Pro, ein Mega S, ein Mega X, ein i3 Mega. Auch diverse User im Forum von drucktipps3d.de haben das bestätigt, nachdem wir sie darauf hingewiesen haben.

Im Anhang zwei Bilder, die ich gerade vorhin an einem komplett unmodifizierten i3 Mega, den ich vorgestern dem Foren-Admin Stephan abgeschwätzt habe, gemacht habe: 17 °C mehr in Realität als eingestellt und angezeigt.

[image]
[image]

Ich bitte darum, diesen Fehler nicht weiterhin nachzubilden sondern zu korrigieren.

knutwurst, 2022-01-21:
Duplicate #207 

Fixed in 7ef2dbd

sorry I haven't looked into it for so long :) 

Best,
Olli

closed 2022-01-21 by knutwurst

alfrankgit, 2022-01-21:
Hallo!

Danke für die Info, habe auch schon lange nicht mehr reingeschaut, da ich die Anycubics abgeschafft habe, d.h. sie stehen in unzugänglichen Ecken.

Fein, dann kann ich ja mal die aktuelle Version testen

Ich würde darüber dann auch gerne einen Bericht auf drucktipps3d.de <http://drucktipps3d.de/> schreiben, ist das OK für Euch?

Oder soll ich noch etwas warten, bis die 1.4 aus der Beta-Phase draußen ist?

Alex.

knutwurst, 2022-01-21:
Nein, du kannst gern schreiben was du magst :) 
Die finale 1.4 ist in Arbeit, aber es gibt noch einiges zu tun und ich weiß nicht, wann ich soweit bin.

alfrankgit, 2022-01-22:
Eine neue Erkenntnis noch:

Der unmodifizierte i3 Mega mit dem weißen Display benötigt Thermistor Typ 1.

Dogfroster, 2022-01-24:
> Eine neue Erkenntnis noch:
> 
> Der unmodifizierte i3 Mega mit dem weißen Display benötigt Thermistor Typ 1.

Also hab den ganz normalen I3 Mega  mit dem Weiss Blauen (erstes Bild von dir)Standarddisplay. 
Welchen brauch ich nun 11 oder 1 ???

knutwurst, 2022-01-24:
Also ich habe zwei Mega S hier (neues und altes Display) und auch Ersatz-Hotends von Anycubic... da stecken überall Thermistoren drin, die von der Kurve her quasi Typ 1 sein könnten. Typ 5 ist tatsächlich auch nah dran - aber mit höherer Grundtemperatur. 

Deswegen ist jetzt ab der 1.4.0 Beta 5 der Default für die MEGA Reihe der Typ 1 und für den 4MAX Pro Typ 11.