knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #269: [BUG] Mesh leveling not

mrpilipo, 2022-01-30:
Running 1.3.1 on Mega S, with 0.0.2 board.
Installed BLTouch 3.1 following instructions from here, downloaded and flashed MEGA_S_BLT_10_v1.3.1, restored defaults, saved to eeprom. BLTouch passes self test and responds to commands as intended, found my z offset also following instructions from here.
Using start and end gcode provided here (M420 S1 after G28, to enable mesh leveling).

When checking my bed level in octoprints bed visualizer I can see manual adjustments being reflected accordingly.

Now for the issue:
Test print results in lines **squished to transparency where bed appears to be low** and **barely sticking where it appears to be high**.

mrpilipo, 2022-01-31:
I might have solved the issue...
In the meantime change firmware to klipper... and the issue persisted...
It is the bed, bed carriage to be exact, even tho I've tightened the sprints as much as possible, it still has some give on the left side (with one support) and bends a tiny bit from the force of the probe...

closed 2022-01-31 by mrpilipo

knutwurst, 2022-01-31:
This is a common issue with the Mega/S/P Printers. Take a look here:
https://habitoti.de/wordpress/bltouch-frustration/

mrpilipo, 2022-01-31:
The more you know... took my tiny bubble level to the carriage and... level visibly changes as it goes from one end of the x axis to the other... time to go back to Marlin and do some math since there is no simple way to edit mesh point in klipper (outside of manually editing them in config file)