knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #273: [BUG] (mesh bed leveling next point activated by default with different item selected on LCD)

weirdgyn, 2022-02-02:
### Bug Description

When using the mesh bed leveling feature activated by LCD menu on Anycubic i3 Mega S (Trigorilla 1.0) the "Next leveling point" command is executed automatically when pushing on the 'refresh' button even if a different option is supposedly selected.

### Steps to Reproduce

1. Access the Mesh Bed Leveling feature
2. Start mesh bed leveling
3. Select the "Next leveling point" at least once
4. Confirm with refresh button
5. [The last selected option on LCD "Up 0.02 mm" should be selected"] Confirm with refresh button.

**Expected behavior:** [What you expect to happen]

The head should move up 0.02 mm.

**Actual behavior:** [What actually happens]

The "Next leveling point" command is executed without any Z movement

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

knutwurst, 2022-02-02:
This is not a bug that I can fix. The problem is that the display also refreshes the list when you use the refresh button. And then the first element is always selected, whether you like it or not.

The firmware works in such a way that the last(!) point always remains selected. So you can click "UP 0.02" and then click refresh multiple times if you like.

But you have to do this once after "Next Mesh Point", otherwise the next mesh point will be selected each time.

Is stupid I know. But unfortunately there is no other way, because the menu consists of a misused file list.

Best,
Olli

weirdgyn, 2022-02-02:
ok

knutwurst, 2022-02-02:
I'm sorry I can't help :/

closed 2022-02-02 by knutwurst

weirdgyn, 2022-02-02:
it was not a big issue anyway but I like to give my contribution if possible..