knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #274: Mega X - loss of connection to OctoPrint

meista-roerich, 2022-02-02:
Hello knutwurst,

a few days ago I updated my Mega X to your latest firmware 1.3.1. 
Since then I thought I had the same problem with the USB-connection Mega X to OctoPrint, that was discribed in another post. 
I disconnected the webcam from the raspi as mentioned in the post, but the problem still occured.
The following error message came up in the webinterface of OctoPrint when the printer wants to start printing:

_The firmware of your printer has reported an error. Therefore OctoPrint has closed the connection. Error reported: Failed to enable Bed Leveling_

Before I wanted to start with the first print after the update, I made all settings as discribed.

I also used the Start- and Endcode you use and inserted it in Cura.
Used Cura version is:
4.13.1

Used version of OctoPrint:
    OctoPrint 1.7.3
    Python 3.7.3
    OctoPi 0.17.0

The firmware I installed on the Mega X is:
MEGA_X_TMC_v1.3.1.hex

What can I do further?
Thanks in advance for your support!

knutwurst, 2022-02-03:
Hi @meista-roerich ;)  (Werner? :p )

The USB connection problem only occurred with activated WLAN. This is only and exclusively active in 1.3.0. Since you have the 1.3.1, this problem does not affect you.

You took my start code, right? ;)
This is not a bug, but a user error (sorry for that).

Your question has come up many times and can be found in the wiki or in the FAQ:

https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/FAQ-(english)#when-i-try-to-start-printing-the-console-shows-failed-to-enable-bed-leveling

meista-roerich, 2022-02-03:
Hello @knutwurst,

thanks for your quick response!
Yes, you're right. In this case the 'Schnüffelstück' was missing😂
No sooner do you do it right than it works 😉. 

Thanks a lot!

closed 2022-02-15 by knutwurst