knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #276: Bed heating very very slowly after update to 1.4.0

fizulnizam, 2022-02-09:
Greetings, 

As the title says, my i3 Mega with 2 Z endstops ultrabase bed heats up super slowly up til the thermal runaway protection kicks in. 

I am heating it up to 110 degrees Celcius for ABS prints and it will just not reach the said temperature.

It worked well before and now even when I downgrade to 1.3.1 I still have the same issue which is pretty weird.

Can't even PID Tune the bed since even when PID tuning, the temp will climb very slowly and thermal runaway will kick in anyways.

How may I solve this?

knutwurst, 2022-02-11:
It's fixed in the 1.4.1 ;)

fizulnizam, 2022-02-12:
Updated to 1.4.1 and the issue still persists at 98 degrees Celcius.. How else may I try to fix this?

knutwurst, 2022-02-12:
110°C with the original Ultrabase is ... ambitious. Without thermal insulation, the power is often not sufficient to reach and hold the temperature.

Try to buy an insulating mat that you stick under the Ultrabase. Cork is ok too. Also do a PID tune to 110°C. This is only possible via USB via Pronterface or another terminal program.

Also check if your screw terminals are tighten and remove the solder from the wiring. It is absolutely necessary that the contact resistance is kept as low as possible.

fizulnizam, 2022-02-12:
I will give that a try but thank you so much for your valuable advice!

closed 2022-02-12 by fizulnizam