knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #279: Mega Pro - Z-limit switch not working correctly, it forces (can force) below 0 after touching the switch [BUG] (short description)

eraxare, 2022-02-16:
### Bug Description

I'm using MEGA_P_DGUS_v1.4.2 on stock Mega Pro, it tried to (can) go further after touching the Z-limit switches when jogging the Z axis, the issue not exist on the X/Y axis/switches. It was almost crashed the glass when first time homing, and homing correctly worked after reset but the issue persist when manually jogging. Can you please fix it?
### Steps to Reproduce

Go to the axis control on the touch menu or Printer controls in Cura and lower the Z axis below 0.0.

whopperg, 2022-02-18:
Well i noticed that too - i initially thought that this was happening due tmc2208 and firmware update stuff i did... but after resetting firmare to default settings, this didnt happen anymore

knutwurst, 2022-02-18:
Please read the FAQ. Your question might be already answered. 
if not, give a complete list of steps to reproduce.

closed 2022-03-27 by github-actions[bot]

zpxp, 2022-03-31:
I get this problem as well. It happens after doing the home and only on the Z axis