knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #280: [BUG] Wrong thermistor default value in configuration.h

i-chebykin, 2022-02-17:
### Bug Description

The default values for the  TEMP_SENSOR_0 and TEMP_SENSOR_BED are `1`
According to the `Marlin\src\module\thermistor\thermistor_1.h` this corresponds to  
  `R25 = 100 kOhm, beta25 = 4092 K, 4.7 kOhm pull-up, bed thermistor`  
But beta of the ai3m printer is B3950 from the stock
Moreover `13` is commented as B3950, but accordint to the `thermistor_1.h`  
the value `13` corresponds to `R25 = 100 kOhm, beta25 = 4100 K, 4.7 kOhm pull-up, Hisens thermistor`

So, it is needed to fix wrong defaults and wrong comments

### Steps to Reproduce

just run default build and measure real temperature

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]
The real temperature is very different from the displayed values

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

knutwurst, 2022-02-20:
I don't quite understand the bug ticket. How do you get 13? Type 1 and Type 11 are installed by Anycubic, although these hardly differ and Type 11 mostly applies to newer printers such as the 4MAX. Type 13 is never used.

But yes, the description is misleading. But it comes directly from the Marlin sources. However, since I don't use them, I don't think I'll change them.

i-chebykin, 2022-02-20:
ok, I see
closed the issue

closed 2022-02-20 by i-chebykin