knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #283: [BUG] M92 from Mega for Mega S

esprudzs, 2022-02-18:
### Bug Description

While compiling for Mega_S_TMC_BLT_10, E steps are set to 98, which seems to be the correct value for Mega instead of 400 (384, actually) for Mega S.

### Steps to Reproduce

1. Compile for Mega_S_TMC_BLT_10
2. Upload to printer
3. Check M92 value

**Expected behavior:** [What you expect to happen]

Should be 400 for Mega S.

**Actual behavior:** [What actually happens]

Is 98, which is the number for Mega.

#### Additional Information

Workaround: Type M92 T0 E400 (384) in the terminal.

While mathematically, 400 is a correct number, stock Mega S is set to 384 - slight extrusion. Setting this value instead of the mathematical 400 will benefit most users and increases 'out of the box' print quality by a lot.

knutwurst, 2022-02-18:
The firmware default ist 393 Steps for the Mega S Titan Extruder which is also loaded, when installing the firmware to the vanilla printer. Maybe you didn't load the fw defaults so ensure, that the EEPROM is initialized?

esprudzs, 2022-02-18:
Well, I took 384 off a just unboxed Mega S to investigate why the print quality dropped when I first installed knutwurst firmware. This is what I found to be the big factor. To check, I just unboxed and connected another one - 384 it is. The configuration.h I'm looking at says 393, there must have been an older version I remember 400 from. Or maybe I'm just wrong.

When it comes to loading fw defaults - you are right, my bad, I must have forgotten to do that, you can close the issu.

closed 2022-02-18 by knutwurst