knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #285: anycubic mega x[BUG] (short description)

ilya83p, 2022-02-19:
In the firmware MEGA_X_TMC_BLT_10_v1.4.2.hex when printing or calibration is turned on, the nozzle rises above the table about 20-22mm, the probe measures through the air and an error occurs. Please tell me how to fix it. Thanks

### Bug Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

knutwurst, 2022-02-20:
That's not a big in the firmware. It's most likely an issue with your wiring. Please double check all connections.

ilya83p, 2022-02-20:
It's not a big deal in firmware.  Most likely the problem is in your wiring.  Please double check all connections.  Trigoilla 0.0.2 connected me to s1 where the pins stick out.  when you turn on the observation, it uses a probe to test everything as it should.  everything according to the instructions.

Отправлено из мобильной Почты Mail.ru

воскресенье,

ilya83p, 2022-02-20:
Or in three gorilla 0.0.2 you need to connect a three-plug wire to another place

ilya83p, 2022-02-20:
I'm just suffering for the second day, everything seems to be fine everywhere, I've done Z-Offset 20 times, and nothing comes out, please tell me, be so kind

ilya83p, 2022-02-20:
Thank you very much, in the boards of version 1.1, the D11 pin is in a different place than in 1.0

closed 2022-03-28 by github-actions[bot]