knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #286: [BUG] Would not be possible without knutwurst on affordable hardware

esprudzs, 2022-02-19:
Mega S, customized for multi material, single nozzle printing, 3DTouch, TMC2008, dual part cooling fan. Thank you!

[image]
[image]

PS Maybe an "I did this with knutwurst" section is due here :)

knutwurst, 2022-02-19:
Many thanks!
I'm glad your printer is working so well :)

drsurfer, 2022-02-27:
I wonder why this post is marked [BUG] and still not closed. I understand the will to praise the effort of Oliver & co., but polluting this section isn't the right way to do. Just my opinion.

closed 2022-02-27 by knutwurst