knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #287: [BUG] (laserstärke - S-wert)

oooBuGooo, 2022-02-19:
Ich habe das Problem dass ich den S wert nicht genau ermitteln kann, um die laserintensität richtig einzustellen.

Was ist der genaue Max. S-Wert für den Laser steuerbefehl M03/M05?

stklcode, 2022-02-20:
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/0440770618e88d3c24db8f65957c7091b302fa02/Marlin/Configuration_adv.h#L3419-L3426

Power Unit ist auf PWM eingestellt (Standard), damit sind theoretisch Werte von 0-255 für [M3](https://marlinfw.org/docs/gcode/M003.html) und [M4](https://marlinfw.org/docs/gcode/M004.html) zulässig (praktisch sind es momentan max. 254, siehe #259). [M5](https://marlinfw.org/docs/gcode/M005.html) hat keine Parameter.

oooBuGooo, 2022-02-20:
Vielen dank für die Lösung.

Ich war nur sehr verunsichert da ich bei dem gleichen Drucker mit original Software einen schwarzen Kasten mit 10% gelasert habe doch ist wahrscheinlich 
und der unterschied war immens.
10% mit der original Software ist bei Lightburn 55% bei ungefähr gleicher Geschwindigkeit.
Wahrscheinlich hat Anycubic direkt ein par Parameter verändert.
Ich habe es jetzt raus und bin sehr glücklich.

closed 2022-02-20 by oooBuGooo

stklcode, 2022-02-21:
Die Implementation von Anycubic skaliert in frühen Version gar nicht (nur an/aus) oder von 0-100. Dabei wird sich nicht auf die Hardware PWM verlassen, sondern eine eigene Zeitschleife aufgebaut. Marlin 1 hat ja standardmäßig so gut wie nichts in Richtung Laser an Board, da musste man kreativ werden. In Marlin 2 verwenden wir jetzt die Standardeinstellung, 2,5kHz Hardware PWM.

Ich bin gerade in den letzten Zügen, das originale BMP Druckfeature zu portieren. Ist nicht so wahnsinnig nützlich, aber damit kommt auch die Indication über das Display und es gibt tiefe Einblicke in das, was da ursprünglich gebaut wurde. Eventuell gibt es in dem Zug noch etwas Feintuning der Laser Parameter.