knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #288: Z-Achse vertauscht - Home klappt nicht

vossi86, 2022-02-20:
Hallo, ich hab mir heute auf den Anycubic i3 Mega Pro die Firmware "MEGA_P_DGUS_TMC_v1.4.2" aufgespielt. Ich war mir sicher, das ich die richtige habe. BL Touch ist nicht verbaut und das Display sieht genau aus wie der Clone.

NAach dem installieren, habe ich wie im Leritfaden angegebn erstmal die Schritte FW Default und EEPROM Save gemacht. Soweit so gut. Danach wollte ich den Drucker in die HOME schicken, dabei ist die Z-Achse soweit runter gefahren, dass sie das Bett komplett nach unten gedrückt hat.

Ich dachte also das ich die falsche Firmware habe. Nochmal nachgeschaut und tatsächlich sollte es die die falsche sein. Ich habe die TMC ja nicht nachträglich verbaut, sondern diese wohl schon drinne gehabt. Also muss ich ja, korriegiert mich wenn ich flasch liege, folgende Firmware nehmen "MEGA_P_DGUS_v1.4.2".

Also diese genommen und aufgespielt. Auch hier wieder die Schritte ausgeführt. Wollte den Drucker in die Home schicken. Jetzt fährt der Drucker auf der Z-Achse aber nach oben anstatt nach unten und versucht da ins unenliche weiter hinauf zu kommen.

Was mache ich den hier flasch?

Liebe Grüße

knutwurst, 2022-02-22:
Das ist zu 99% eine falsche Verkabelung. Der Sensor löst aus, bevor er das Ende erreicht. Also wurde vermutlich das schwarze und weiße Kabel vertauscht.

vossi86, 2022-02-22:
Hey, danke für die Rückmeldung. Müsste das dann nicht aber auch mit der Originalen FW auch passieren?

Ich werde es nochmal versuchen zu reproduzieren und schauen was ich falsch gemacht haben könnte.

knutwurst, 2022-02-22:
Oh Sorry! Ich hab gedacht, du hättest einen BLTouch Sensor verbaut! Mein Fehler.. tut mir leid. Das kommt davon, wenn man unterwegs mit dem Handy liest :D 

Also ja, installiere doch noch mal die Originalfirmware und schaue dir das Verhalten an. Es sollte eigentlich keinen Unterschied machen, ob du die OFW oder die MEGA_P_DGUS installiert hast. Die Laufrichtung ist dieselbe.

Sollte das Problem weiterhin bestehen, versuche mal bitte die Version 1.3.1. Wenn die dann geht, gibt es einen Bug. Allerdings wärst du der erste, bei dem das so ist :/ 

lg
Olli

vossi86, 2022-02-22:
Werd ich testen und gebe Rückmeldung. Danke

vossi86, 2022-02-22:
So, nachdem ich heute hiem gekommen bin, hab ich mich nochmal dran gemacht und die FW 1.4.3 installiert. Es hat geklappt. Ich hab diesesmal alles genauso gemacht wie beim letzten mal, nur deinen GCode habe ich weggelassen (ich denke daran wird es nicht gelegen haben). Easy leveling klappt, PID tuning läuft aktuell. Mesh teste ich auch noch. Also bisher scheint es zu laufen ohne Probleme zu machen. Ich weiß nicht was, aber ich hab wohl beim ersten mal was falsch gemacht.

Ich schließe dann mal und danke für die Hilfe!

closed 2022-02-22 by vossi86