knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #289: [BUG] Ticking sound after start of print

v-a-d-e-r, 2022-02-20:
I get still ticking sound after a couple of seconds and the printer stops.
Printer: Anycubic Mega S
FW: v1.4.3
Can this issue confirmed by others?

knutwurst, 2022-02-21:
Please post the log output :)

v-a-d-e-r, 2022-02-21:
What should be in the log? Annoying sound on and crash immediately... *g*

knutwurst, 2022-02-21:
Uhm... if you don't post a log like mentioned in the FAQ, I can't help you because nobody will know what's going on.

v-a-d-e-r, 2022-02-21:
Hmm, I'm not sure if that output from cura (4.13.1) really helps....

2022-02-21 14:42:22,863 - DEBUG - [MainThread] SliceInfoPlugin.SliceInfo._onWriteStarted [118]: 'info/send_slice_info' is turned off.
2022-02-21 14:42:22,865 - DEBUG - [MainThread] ArcWelderPlugin.ArcWelderPlugin._filterGcode [252]: Using temporary file /tmp/tmpvrui7u2f
2022-02-21 14:42:22,869 - DEBUG - [MainThread] ArcWelderPlugin.ArcWelderPlugin._filterGcode [282]: Running ArcWelder with the following options: ['/home/kingpin/.local/share/cura/4.13/plugins/ArcWelderPlugin/ArcWelderPlugin/bin/linux/ArcWelder', '-m=1000000.000000', '-t=0.050000', '-r=0.050000', '-d', '-g', '/tmp/tmpvrui7u2f']
2022-02-21 14:42:23,184 - DEBUG - [MainThread] ArcWelderPlugin.ArcWelderPlugin._filterGcode [296]: 2022-02-21 14:42:22.015 - arc_welder.gcode_conversion - INFO - Source and target path are the same.  The source file will be overwritten.  Temporary file path: /tmp/76931fac-9dab-2b36-c248-b87d6ae33f9a.tmp
2022-02-21 14:42:22.015 - arc_welder.gcode_conversion - INFO - Processing Gcode
	Source File Path             : /tmp/tmpvrui7u2f
	Target File Path (overwrite) : /tmp/76931fac-9dab-2b36-c248-b87d6ae33f9a.tmp
	Temporary File Path          : /tmp/76931fac-9dab-2b36-c248-b87d6ae33f9a.tmp
	Resolution                   : 0.0500mm (+-0.02500mm)
	Path Tolerance               : 5.000%
	Maximum Arc Radius           : 1000000mm
	Min Arc Segments             : 0
	MM Per Arc Segment           : 0.000
	Allow 3D Arcs                : False
	Allow Dynamic Precision      : True
	Default XYZ Precision        : 3
	Default E Precision          : 5
	G90/G91 Influences Extruder  : True
	Log Level                    : INFO
	Hide Progress Updates        : False
2022-02-21 14:42:22.017 - arc_welder.gcode_conversion - INFO - arc_welder::process - Parameters received: source_file_path: '/tmp/tmpvrui7u2f', target_file_path:'/tmp/76931fac-9dab-2b36-c248-b87d6ae33f9a.tmp', resolution_mm:0.05mm (+-0.03mm), path_tolerance_percent: 0.05, max_radius_mm:1000000.00, min_arc_segments:0, mm_per_arc_segment:0, g90_91_influences_extruder: True, allow_3d_arcs: False, allow_dynamic_precision: True, default_xyz_precision: 3, default_e_precision: 5
Progress:  percent_complete:0.00, seconds_elapsed:0.00, seconds_remaining:inf, gcodes_processed: 0, current_file_line: 0, points_compressed: 0, arcs_created: 0, num_firmware_compensations: 0, compression_ratio: 0.00, size_reduction: 0.00% 
Progress:  percent_complete:100.00, seconds_elapsed:0.28, seconds_remaining:0.00, gcodes_processed: 85315, current_file_line: 86506, points_compressed: 32231, arcs_created: 6222, num_firmware_compensations: 0, compression_ratio: 1.46, size_reduction: 31.29% 
2022-02-21 14:42:23.296 - arc_welder.gcode_conversion - INFO - Deleting the original source file at '/tmp/tmpvrui7u2f'.
2022-02-21 14:42:23.297 - arc_welder.gcode_conversion - INFO - Renaming temporary file at '/tmp/76931fac-9dab-2b36-c248-b87d6ae33f9a.tmp' to '/tmp/tmpvrui7u2f'.
2022-02-21 14:42:23.297 - arc_welder.gcode_conversion - INFO - 
   Min          Max     Source  Target  Change
----------------------------------------------
  0.000mm to   0.002mm       0       0    0.0%
  0.002mm to   0.005mm       2       2    0.0%
  0.005mm to   0.010mm      25      14  -44.0%
  0.010mm to   0.050mm     230     110  -52.2%
  0.050mm to   0.100mm    1195     215  -82.0%
  0.100mm to   0.500mm    8756    3543  -59.5%
  0.500mm to   1.000mm   15867    5530  -65.1%
  1.000mm to   5.000mm   24543   14122  -42.5%
  5.000mm to  10.000mm     328     826  151.8%
 10.000mm to  20.000mm     160     509  218.1%
 20.000mm to  50.000mm      16     150  837.5%
 50.000mm to 100.000mm       1      93 9200.0%
          >= 100.000mm       0       0    0.0%
----------------------------------------------
Total distance source:.............53571.225mm
Total distance target:.............53575.154mm
   Total count source:...................51123
   Total count target:...................25114
 Total percent change:..................-50.9%
2022-02-21 14:42:23.297 - arc_welder.gcode_conversion - INFO - Arc Welder process completed successfully.

2022-02-21 14:42:23,203 - DEBUG - [MainThread] UM.Controller.setActiveStage [180]: Setting active stage to MonitorStage
2022-02-21 14:42:23,208 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [424]: file:///tmp/.mount_UltimatO8aec/usr/bin/resources/qml/PrintSetupSelector/Custom/QualitiesWithIntentMenu.qml:168:25: QML Binding: Not restoring previous value because restoreMode has not been set.
This behavior is deprecated.
You have to import QtQml 2.15 after any QtQuick imports and set
the restoreMode of the binding to fix this warning.
In Qt < 6.0 the default is Binding.RestoreBinding.
In Qt >= 6.0 the default is Binding.RestoreBindingOrValue.

2022-02-21 14:42:26,225 - DEBUG - [MainThread] UM.Controller.setActiveStage [180]: Setting active stage to PrepareStage
2022-02-21 14:42:26,242 - DEBUG - [MainThread] cura.Machines.Models.NozzleModel._update [28]: Updating NozzleModel.
2022-02-21 14:42:36,562 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.006197452545166016 seconds
2022-02-21 14:48:24,272 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [48]: Scheduled the next on-app-exit callback [<bound method USBPrinterOutputDevice._checkActivePrintingUponAppExit of <USBPrinting.USBPrinterOutputDevice.USBPrinterOutputDevice object at 0x56480 [... cut ...]
This behavior is deprecated.
You have to import QtQml 2.15 after any QtQuick imports and set
the restoreMode of the binding to fix this warning.
In Qt < 6.0 the default is Binding.RestoreBinding.
In Qt >= 6.0 the default is Binding.RestoreBindingOrValue.

2022-02-21 14:48:26,171 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [424]: file:///tmp/.mount_UltimatO8aec/usr/bin/qml/UM/Preferences/PreferencesDialog.qml:113: TypeError: Cannot read property 'resetEnabled' of null

 *** CRASH ***

v-a-d-e-r, 2022-02-21:
This is what Pronterface logs before the printer died:

SENT: M105
RECV: ok T:178.77 /195.00 B:48.65 /45.00 @:127 B@:0
SENT: M105
RECV: ok T:177.79 /195.00 B:48.40 /45.00 @:127 B@:0
SENT: M105
RECV: ok T:177.52 /195.00 B:48.17 /45.00 @:127 B@:0
RECV: Error:Thermal Runaway, system stopped! Heater_ID: E0
Error:Thermal Runaway, system stopped! Heater_ID: E0

SENT: M105
RECV: echo:THERMAL RUNAWAY
RECV: Error:Printer halted. kill() called!
Error:Printer halted. kill() called!

RECV: //action:poweroff

v-a-d-e-r, 2022-02-21:
PID tuning works fine for the heat bed and the hot end. No issue there! Component cooler running at only 60%

knutwurst, 2022-02-22:
Heater E0 is the hot end. And since there isn't a much too high temperature error (which would occur with a thermistor error) I'm guessing a broken wire or a burnt wire terminal for the hotend cartridge heater.

The Thermal Runaway Protection is a safety feature. And there is something wrong with your printer.

If this only occurs when printing or sporadically, it must be a broken cable or a burnt cable terminal. Check all green plug and screw connections. Even inside the printer!

v-a-d-e-r, 2022-02-22:
That's all I know. When I set the temp again after the drop via display, the printer works like expected. It seems also to be ever at the same position of the print. Same effect with an other model. So it can not be an hardware issue.

knutwurst, 2022-02-22:
Did you change anything else on the printer? Fan? fan duct? How is the fan speed set in the slicer?

The problem must be there somewhere and these errors are also known and therefore almost all dealt with in the FAQ :)

v-a-d-e-r, 2022-02-22:
There is close to nothing original at the printer. ;-) I have no clue what happend in the past few days. It worked also perfect with the beta versions of 1.4.2. The slicer was not changed and no change in the settings there. The only thing I changed was your FW. Very curious.....

esprudzs, 2022-02-22:
Well, I installed dual fan duct and experienced the same problem. Not on every print and not always, but sometimes mid 10-hour print. I ended up increasing the time firmware would wait for the temp to come to expected in the config file to prevent the problem. It is around line 306 in configuration.h:
     #define WATCH_TEMP_PERIOD 60                // Seconds

This solved it for me for good.

v-a-d-e-r, 2022-02-23:
Ah, ok. But I do not self-compile. Maybe it's a hint for @knutwurst

closed 2022-03-31 by github-actions[bot]