knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #290: [FR] TMC Optimisations for instances where driver wires are swapped

MrBill1234, 2022-02-21:
### Description

Documentation says:

https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Beginner's-Guide-(English)#4-download-the-correct-firmware

_**_TMC stands for Trinamic TMC motor driver. The direction of rotation of the motors is inverted and the control of the motors for TMC drivers is optimized.**_

It is unclear if the non _TMC firmware contains these TMC optimisations for when you are using a TMC driver, but have manually inverted the wires for the driver.

### Feature Workflow

None

#### Additional Information

If these optimisations are of use, would it be possible to get a _TMC for when the wires have been manually changed/swapped on the board?   

What exactly are the TMC optimisations anyhow?

knutwurst, 2022-02-21:
Hi,
the difference is only the difference in the minimum delay before and after setting the stepper direction.

A4899: 200ns (pre- and post delay)
TMC22xx: 20ns pre and 10ns post delay.

So the TMC drivers are a bit more responsive when running in TMC mode.

But: TMC can also work in legacy mode with A4988 timings so there is no need to turn the cables back if you use the non-TMC Version.

Best,
Olli

MrBill1234, 2022-02-21:
Thanks for the response.  Any disadvantage to running in legacy mode, or advantage to having the better response?  To be honest, I installed them (several years ago) for the silence - and they are still doing this well!  But if I can get a small edge on quality or some other advantage by not running with legacy mode timings - why not :)

knutwurst, 2022-02-21:
To be honest I can't answer the question. I don't see any difference myself, but I have the feeling that the drivers are a bit louder in legacy mode. But it can also be imagination.

On the other hand: Never change a running system :D

MrBill1234, 2022-02-21:
Right - if it isn't broken, don't fix it ;)

closed 2022-03-29 by github-actions[bot]