knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #291: [BUG] Z-offset on Chiron is not correct after auto leveling

JonasGe, 2022-02-21:
### Bug Description

When running the auto leveling with the stock probe, the Z-offset is not correct (nozzle is too high).
When I add a M420 S1 L1 to the start GCode, the nozzle crashes into the bed.

### Steps to Reproduce

1. Install FW
2. Restore default EEPROM values & save values
3. Reset mesh
4. Auto leveling with stock probe, finishes probing all points and beeps.
5. Save values in EEPROM
6. Start GCode listed on the wiki of this repo, nozzle prints in the air

**Expected behavior:** Auto-level and press print for a good quality print

**Actual behavior:** Auto-level, press print and produce plastic spaghetti.

#### Additional Information

If I add the Z-offset to the start GCode with `M851 Z-XX.XX` and M420 S1 L1, the printer works fine.
I believe you removed the M420 part from the start GCode in the wiki, was there a reason to remove it?

knutwurst, 2022-02-21:
What firmware version are you using? M420 S1 is no longer necessary as the mesh is (or should be) automatically activated immediately after the G28.

If it doesn't, then it's a bug :/

JonasGe, 2022-02-21:
Your latest release, 1.4.3. I'll try with removing M420 and leaving the M851.

knutwurst, 2022-02-21:
Also the Offset should be saved when changed via the touchscreen. Strange.

knutwurst, 2022-02-21:
Damn.. I'll check this. Thanks!

JonasGe, 2022-02-21:
I'll paste my M503 output when my print finishes.

closed 2022-03-29 by github-actions[bot]