knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #293: [BUG] Thermistor-Type

chaesselbarth, 2022-02-22:
### Bug Description

I assume that the configured thermistor type is wrong.

### Steps to Reproduce
See below and refered commits. 

**Expected behavior:** [What you expect to happen]

For all stock anycubic printers the thermistor type should be configured to 11. See [this commit](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/commit/7ef2dbdac30df099f63fb4c831f686988d69615b).

**Actual behavior:** [What actually happens]
Due to commit [0d4346542b79469cd5f8b5a3a825d9aa0812ab82](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/commit/0d4346542b79469cd5f8b5a3a825d9aa0812ab82) the thermistor type is set 1 for all stock anycubic printers.

#### Additional Information

None

knutwurst, 2022-02-23:
I originally thought Type 11 would be the right one too, but most Anycubic printers use Type 1. However, this is also very close to the Type 11 curve, so it makes little difference.

Type 1 is correct for my printers. So far I haven't found one where type 11 fits better. For the 4MAX I relied on the ticket you linked.

chaesselbarth, 2022-02-24:
ok, thanks. Now it's clear to me. Sorry for opening the issue, issue will be closed...

closed 2022-02-24 by chaesselbarth