knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #294: [FR] (Release Wifi Version of Firmware)

OwenD30, 2022-02-23:
### Description

Could you release a wifi version of each new firmware release so those of us who want to use it don't have to compile it ourselves? Or make it togglable on/off on the special menu?

### Feature Workflow

1. Wifi version released or menu modified to have wifi control
2. That's it

#### Additional Information

I downloaded 1.4.0 on my Mega that was using 1.3.0 previously with the ESP3D, and I realized the wifi feature had been disabled. It would be great if a wifi version was released alongside each new update because a lot of us don't want to have to recompile the firmware to include wifi. Or, maybe, you could have wifi control in the special menu that allows it to be toggled on/off, similar to the filament runout sensor.

P.S. Your firmware is awesome, thanks so much for all the work you do!

knutwurst, 2022-02-23:
Nope, sorry :)
There are only a few people who actually use this feature and I already have 60 assets. A Wifi version would double the number.

Compiling the firmware by yourself is not as hard as you might think. Currently I only have detailed instructions in german, but I'll translate them by time. 

See: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Howto:-Firmware-mit-PlatformIO-selbst-kompilieren

OwenD30, 2022-02-24:
Ok wow I had no idea compiling was that easy. I just used Google to translate the page to English. Thanks so much!

OwenD30, 2022-02-24:
Really quick, should I also uncomment #define BAUDRATE_2 or just the #define SERIAL_PORT_2?

knutwurst, 2022-02-24:
Depends... You can use whatever you like. If you want the full 500k speed, you need to set BAUDRATE_2 to 500.000. Since file transfer is slow af, it's also totally usable with 250k or lower.

OwenD30, 2022-02-25:
Ok thanks!

closed 2022-02-25 by OwenD30