knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #295: Compilation of "MEGA_P_DGUS_BLT_10" fails with errors (Vanilla)

fully-sikh, 2022-02-24:
### Bug Description

I am following instructions from FAQ on how to compile your firmware.
Downloaded a fresh copy of the code @ 7pm 24/02/2022

### Steps to Reproduce

1. Download a fresh copy of the code from github
2. Unzip to folder called "Knutwurst"
3. Open folder in VSCode
4. Click on PlatformIO--> Project Tasks--> MEGA_PDGUS_BLT_10-->Build
5. Error message while compiling

**Expected behavior:** Code should compile and produce a .HEX file

**Actual behavior:** Error messages while compiling

#### Additional Information

Compiling .pio\build\MEGA_P_DGUS_BLT_10\src\src\module\tool_change.cpp.o
Marlin\src\module\configuration_store.cpp: In static member function 'static void MarlinSettings::postprocess()':
Marlin\src\module\configuration_store.cpp:434:12: error: 'i' was not declared in this scope
   LOOP_XYZ(i) {
            ^
Marlin\src\module\configuration_store.cpp:434:3: error: 'LOOP_XYZ' was not declared in this scope
   LOOP_XYZ(i) {
   ^~~~~~~~
Marlin\src\module\configuration_store.cpp:434:3: note: suggested alternative: 'LOOP_L_N'
   LOOP_XYZ(i) {
   ^~~~~~~~
   LOOP_L_N
Marlin\src\module\configuration_store.cpp: In static member function 'static bool MarlinSettings::_load()':
Marlin\src\module\configuration_store.cpp:1428:38: error: 'i' was not declared in this scope
         if (!validating) LOOP_XYZE_N(i) {
                                      ^
Marlin\src\module\configuration_store.cpp:1428:26: error: 'LOOP_XYZE_N' was not declared in this scope
         if (!validating) LOOP_XYZE_N(i) {
                          ^~~~~~~~~~~
Marlin\src\module\configuration_store.cpp:1428:26: note: suggested alternative: 'LOOP_S_LE_N'
         if (!validating) LOOP_XYZE_N(i) {
                          ^~~~~~~~~~~
                          LOOP_S_LE_N
Marlin\src\module\configuration_store.cpp: In static member function 'static void MarlinSettings::reset()':
Marlin\src\module\configuration_store.cpp:2418:15: error: 'i' was not declared in this scope
   LOOP_XYZE_N(i) {
               ^
compilation terminated due to -fmax-errors=5.
*** [.pio\build\MEGA_P_DGUS_BLT_10\src\src\module\configuration_store.cpp.o] Error 1
================================================================================= [FAILED] Took 22.39 seconds =================================================================================
Environment         Status    Duration
------------------  --------  ------------
MEGA_P_DGUS_BLT_10  FAILED    00:00:22.390

fully-sikh, 2022-03-27:
Hi Champ,
Just commenting so that the issue does not automatically get closed.
Thank you

knutwurst, 2022-03-29:
Hey, sorry for the late answer...
Do you use Windows for compiling? Does the problem persist if you try something like the plain MEGA environment?

fully-sikh, 2022-03-30:
Hi,
I'm using Visual Studio Code on Windows 10.
I just tried compiling multiple environments: MEGA_P_DGUS, MESA_S_DGUS, MEGA_S and MEGA all failed to compile.
Below is part of the error messages from compiling the plain MEGA environment:

========================================================
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\font_size_t.cpp.o
Marlin\src\lcd\HardwareSerial.cpp: In function 'void __vector_54()':
Marlin\src\lcd\HardwareSerial.cpp:92:19: warning: unused variable 'c' [-Wunused-variable]
     unsigned char c = UDR3;
                   ^
Marlin\src\lcd\HardwareSerial.cpp: In member function 'void HardwareSerialClass::begin(long unsigned int, byte)':
Marlin\src\lcd\HardwareSerial.cpp:185:11: warning: unused variable 'current_config' [-Wunused-variable]
   uint8_t current_config;
           ^~~~~~~~~~~~~~
Marlin\src\lcd\HardwareSerial.cpp: In member function 'virtual size_t HardwareSerialClass::write(uint8_t)':
Marlin\src\lcd\HardwareSerial.cpp:292:12: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   while (i == _tx_buffer->tail)
          ~~^~~~~~~~~~~~~~~~~~~
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\standard_char_set.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\unicode.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\ftdi_eve_lib\extended\unicode\western_char_set.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\language\language.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\marlin_events.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\about_screen.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\advanced_settings_menu.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\alert_dialog_box.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\backlash_compensation_screen.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\base_numeric_adjustment_screen.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\base_screen.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\bio_advanced_settings.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\bio_confirm_home_e.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\bio_confirm_home_xyz.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\bio_main_menu.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\bio_printing_dialog_box.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\bio_status_screen.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\bio_tune_menu.cpp.o
In file included from Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\language\language.h:23:0,
                 from Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\language\language.cpp:25:
Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\language\language_en.h:35:0: warning: "SUPERSCRIPT_TWO" redefined
   #define SUPERSCRIPT_TWO   u8"^2"

In file included from d:\3dprinter\knutwurst\marlin\src\inc/MarlinConfig.h:53:0,
                 from d:\3dprinter\knutwurst\marlin\src\marlincore.h:24,
                 from Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\language\language.cpp:23:
d:\3dprinter\knutwurst\marlin\src\core\language.h:558:0: note: this is the location of the previous definition
 #define SUPERSCRIPT_TWO   TERN(NOT_EXTENDED_ISO10646_1_5X7, "^2", "²")

Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\boot_screen.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\case_light_screen.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\change_filament_screen.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\confirm_abort_print_dialog_box.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\confirm_auto_calibration_dialog_box.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\confirm_erase_flash_dialog_box.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\confirm_start_print_dialog_box.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\confirm_user_request_alert_box.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\default_acceleration_screen.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\developer_menu.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\dialog_box_base_class.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\display_tuning_screen.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\endstop_state_screen.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\feedrate_percent_screen.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\filament_menu.cpp.o
Compiling .pio\build\MEGA\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\filament_runout_screen.cpp.o
[... cut ...]
================================================================ [FAILED] Took 22.08 seconds ================================================================

Environment    Status    Duration
-------------  --------  ------------
MEGA           FAILED    00:00:22.076
=========================================================== 1 failed, 0 succeeded in 00:00:22.076 =========================================================== 
The terminal process "C:\Users\DeepVansh\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'MEGA'" terminated with exit code: 1.
===============================================
Thank you in advance champ

knutwurst, 2022-03-30:
Please open the i3_mega_envs.ini and add this to your configuration:
`build_unflags = -Werror=all`

So "-Werror=all" is also in the list of build_unflags.

fully-sikh, 2022-03-30:
"i3_mega_envs.ini" is not available in the INI folder (latest pull from github), I added the  "-Werror=all" into couple of environments (MEGA. MEGA_P_DGUS, MEGA_P_DGUS_BLT10) within file "i3_mega.ini" in the INI folder.

The build succeeded for all three environments.

Thank you for resolving the issue.

closed 2022-03-30 by fully-sikh