knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #301: Compilation of MEGA_S_TMC fails

Alkerion, 2022-03-07:
### Bug Description

I've followed instructions on how to compile Marlin firmware : https://marlinfw.org/docs/basics/install.html

### Steps to Reproduce

Download fresh copy of the 1.4.3 source code from github
Unziped to folder "Marlin-2-0-x-Anycubic-i3-MEGA-S-1.4.3"
Downloaded Configurations-release-2.0.9.2.zip
Copied (overwritten) files : Configuration.h & Configuration_adv.h
From : Configurations-release-2.0.9.2.zip\Configurations-release-2.0.9.2\config\examples\AnyCubic\i3 Mega
To : ...\Marlin-2-0-x-Anycubic-i3-MEGA-S-1.4.3\Marlin
Installed VSCode
Installed AutoBuild Marlin
Installed PlatformIO
Open project folder in VSCode
Clicked on PlatformIO --> Project Tasks --> MEGA_S_TMC --> General --> Build

**Expected behavior:** Code should compile and produce a .HEX file

**Actual behavior:** Error messages while compiling

Marlin\src\MarlinCore.cpp: In function 'void event_filament_runout()':
Marlin\src\MarlinCore.cpp:424:63: error: cannot convert 'const char*' to 'PromptReason' for argument '1' to 'void host_action_prompt_begin(PromptReason, const char*, char)'
       host_action_prompt_begin(PSTR("FilamentRunout T"), false);

#### Additional Information

I've planned to change the temp sensor for a 104NT-4-R025H42G, reason why I need to modify (#Define_Temp_sensor_0 13) and compile FW.
First I've tried not to change anything but to compile untouched FW --> without success
My Anycubic mega S has Trigorilla 1.1 with TMC 2209 with your MEGA_S_TMC_v1.4.3.hex FW.
Not clear to me if I've to change settings in Configuration.h or not before compile.
like TMC2209_STANDALONE instead of TMC2208_STANDALONE

knutwurst, 2022-03-07:
This is a hybrid firmware for all anycubic printers - not compatible with plain marlin. You can not use marlin configuration files.

Sadly the detailed manual is currently available in german: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki

Alkerion, 2022-03-07:
Sorry but I don't understand.
I've downloaded Configurations-release-2.0.9.2.zip and replaced the config files for Anycubic Mega S, so it should works for me.
If not, where can I find the config files you've used to compile the FW ?

knutwurst, 2022-03-07:
You can not download and copy marlin configurations! 
You will break everything this firmware relies on.

Please read the tutorial: https://github-com.translate.goog/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Howto:-Firmware-mit-PlatformIO-selbst-kompilieren?_x_tr_sl=de&_x_tr_tl=en&_x_tr_hl=en&_x_tr_pto=wapp

Alkerion, 2022-03-07:
Thanks for support, I've restarted from scratch and I've finally been able to compile it.
But I've lot of problems, normal ?
[image]
?

knutwurst, 2022-03-25:
Yes. That's totally fine.

OwenD30, 2022-04-04:
> This is a hybrid firmware for all anycubic printers - not compatible with plain marlin. You can not use marlin configuration files.
> 
> Sadly the detailed manual is currently available in german: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki

Just as a helpful note, if you right click on the page you can select "translate to English" and Google will translate the page. I did that and was able to follow the instructions perfectly.

closed 2022-05-11 by github-actions[bot]