knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #305: Extruder Steps issue on Mega-S

sigsegv0x0b, 2022-03-11:
I flashed my Mega-S with 
`
Send: M115
Recv: FIRMWARE_NAME:Marlin 2.0.x (knutwurst, Github) (Feb 18 2022 13:32:11) SOURCE_CODE_URL:github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S PROTOCOL_VERSION:1.0 MACHINE_TYPE:Knutwurst's Anycubic Printer EXTRUDER_COUNT:1 UUID:b6153152-86fb-4cd8-bda7-91c6a04c467d`

When I set the extruder Esteps anywhere near the right value for Mega-S it extrudes a crazy amount of filament, I had to set esteps to 105 for it to extude the right amount.

I can't find why it's doing this, the original Mega-S firmware had it 384

[image]
[image]

Here my current M503
`
Send: M503
Recv: echo:; Linear Units:
Recv:   G21 ; (mm)
Recv: echo:; Temperature Units:
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:; Filament settings (Disabled):
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E105.00
Recv: echo:; Max feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z6.00 E60.00
Recv: echo:; Max Acceleration (units/s2):
Recv: echo:  M201 X3000.00 Y2000.00 Z60.00 E10000.00
Recv: echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
Recv: echo:  M204 P3000.00 R3000.00 T3000.00
Recv: echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>):
Recv: echo:  M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z0.40 E5.00
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Mesh Bed Leveling:
Recv: echo:  M420 S1 Z2.00 ; Leveling ON
Recv: echo:  G29 S3 I0 J0 Z0.08800
Recv: echo:  G29 S3 I1 J0 Z0.50500
Recv: echo:  G29 S3 I2 J0 Z0.57500
Recv: echo:  G29 S3 I3 J0 Z1.21000
Recv: echo:  G29 S3 I4 J0 Z1.19000
Recv: echo:  G29 S3 I0 J1 Z0.18500
Recv: echo:  G29 S3 I1 J1 Z0.41500
Recv: echo:  G29 S3 I2 J1 Z0.61500
Recv: echo:  G29 S3 I3 J1 Z0.76500
Recv: echo:  G29 S3 I4 J1 Z0.88500
Recv: echo:  G29 S3 I0 J2 Z0.34500
Recv: echo:  G29 S3 I1 J2 Z0.55000
Recv: echo:  G29 S3 I2 J2 Z0.66500
Recv: echo:  G29 S3 I3 J2 Z0.78000
Recv: echo:  G29 S3 I4 J2 Z0.78500
Recv: echo:  G29 S3 I0 J3 Z0.44700
Recv: echo:  G29 S3 I1 J3 Z0.60500
Recv: echo:  G29 S3 I2 J3 Z0.72000
Recv: echo:  G29 S3 I3 J3 Z0.80000
Recv: echo:  G29 S3 I4 J3 Z0.78400
Recv: echo:  G29 S3 I0 J4 Z0.54500
Recv: echo:  G29 S3 I1 J4 Z0.65500
Recv: echo:  G29 S3 I2 J4 Z0.74000
Recv: echo:  G29 S3 I3 J4 Z0.78000
Recv: echo:  G29 S3 I4 J4 Z0.75000
Recv: echo:  G29 S4 Z0.00000
Recv: echo:; Endstop adjustment:
Recv: echo:  M666 Z0.00
Recv: echo:; Hotend PID:
Recv: echo:  M301 P20.02 I0.91 D80.76
Recv: echo:; Bed PID:
Recv: echo:echo:  M304 P251.78 I49.57 D319.73
Recv: echo:; Linear Advance:
Recv: echo:  M900 K0.00
Recv: echo:; Filament load/unload:
Recv: echo:  M603 L538.00 U555.00 ; (mm)
`

sigsegv0x0b, 2022-03-16:
I flashed marlin 1.1.9 firmware and esteps work correcly at 384, flashing back with this needs to be set to 108. Don't understand marlin well enough to see why this is the case.  

I have TMC2208 installed without reversed connectors, so the 1.1.9 firmware in question is this one:
https://github.com/davidramiro/Marlin-AI3M/releases 1.4.9

Alexander-L000, 2022-04-01:
I had the same problems as you. (Anycubic Mega i3 S)
These values have improved the flow for me.
(for noobies like me: pronterface.exe -> connect to your Printer Port:COMX @ 250000)
M92 EXXX.XX (in my printer 396.00) (XXX.XX = your value)
M203 E30
M204 R1500.00
M500

closed 2022-05-08 by github-actions[bot]