knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #306: Erweiterung der Lasergravurfläche beim Anycubic Mega Pro[FR] (feature request title)

Patzomat, 2022-03-16:
Hallo Knutwurst,

die Lasergravurfläche ist beim Mega Pro in der Y-Achse ja auf 140mm beschränkt anstatt orginal 210mm.
Gibt es eine Möglichkeit diese Beschränkung in der originalen Firmeware aufzuheben?

Beste Grüße
Patzomat

oooBuGooo, 2022-03-16:
In der original Firmware kannst du es leider nicht umstellen.
Mit Knutwurst und Lightburn kannst du die komplette Fläche vom Druckbett nutzen.

Patzomat, 2022-03-16:
Kann man dann auch Bilder bmps Lasern.
Ich hatte da was gelesen, dass man dann nur die Bewegung das Lasers und die Intensität per gcode steuern kann?!

oooBuGooo, 2022-03-17:
Ja man kann auch Bilder gravieren doch gibt es da momentan noch eine kleine Einschränkung.
Der Laser verringert kurz die Geschwindigkeit wenn er den Laser an und ausschaltet.
Dadurch dauert das gravieren sehr lange.
Ausschneiden funktioniert aber super.

Patzomat, 2022-03-17:
Ok dann muss ich mir das nochmal anschauen.
Danke!

oooBuGooo, 2022-03-17:
Sehr gerne

oooBuGooo, 2022-04-06:
@Patzomat Es währe super wenn du den Chat schließt wenn dein Problem gelöst ist. :-)

closed 2022-04-06 by Patzomat