knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #307: TMC26XStepper library not available

GithubPatrick91, 2022-03-18:
### Bug Description

The used library cannot be installed when build the project, due to 404 status.
`TMC26XStepper=https://github.com/trinamic/TMC26XStepper/archive/master.zip`

### Steps to Reproduce

build any project which uses this lib in this library like:
```
[env:MEGA_TMC_BLT_10]
platform          = atmelavr
board             = megaatmega2560
board_build.f_cpu = 16000000L
lib_deps          = ${common.lib_deps}
    TMC26XStepper=https://github.com/trinamic/TMC26XStepper/archive/master.zip
src_filter        = ${common.default_src_filter} +<src/HAL/AVR>
build_flags       = ${common.build_flags} -DMOTHERBOARD=BOARD_TRIGORILLA_14 -DKNUTWURST_MEGA -DKNUTWURST_TMC -DKNUTWURST_BLTOUCH
build_unflags     = -DKNUTWURST_MEGA_S -DKNUTWURST_MEGA_X -DKNUTWURST_MEGA_P -DKNUTWURST_DGUS2_TFT -DKNUTWURST_ONE_Z_ENDSTOP -DKNUTWURST_CHIRON -DKNUTWURST_4MAXP2 
```

#### Additional Information

i cannot find any repo of [trinamics ](https://github.com/trinamic?tab=repositories) like TMC26XStepper

PIO Error output:
```
Library Manager: Installing https://github.com/trinamic/TMC26XStepper/archive/master.zip
PackageException: Got the unrecognized status code '404' when downloaded https://github.com/trinamic/TMC26XStepper/archive/master.zip:
```

knutwurst, 2022-03-18:
Wow... seems like Trinamic has deleted the repo :( 
I just looked in my source code directory and found the library. I'm uploading it here ([TMC26XStepper.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/8306697/TMC26XStepper.zip)), but to be honest it's not needed anymore. It can just as well be replaced by Lib_TMCStepper.

I just committed the change. It should now build through without errors.

GithubPatrick91, 2022-03-18:
Wow, thank you for your quick fix and for sharing the library!
I did some fixed by myself, thats why i forked it. As i pulled it, there were no includes of this library anymore... and i just went in here to delete my issue, cause i thought i became crazy. :)

Again thanks for your fast action and for your awesome repo!

closed 2022-03-19 by GithubPatrick91

Elektrospeedy, 2022-03-28:
Hallo @knutwurst, 

ich finde echt geil was du machst, deine Beschreibungen sind echt super!

ich bin neu hier und ein Anfänger, ich habe ein Anycubic i3 Mega Pro, ich bin auf deine Firmware gestoßen und wollte es auf jeden fall ausprobieren! Ich habe aber das selbe Problem, wie oben beschrieben, die Bibliothek ist nicht vorhanden, du schreibst das diese mit "Lib_TMCStepper" ersetzt werden soll, wo macht man das? Wird diese im PlatformIO installiert oder muss das im Code abgeändert werden?

Vielleicht hast du dazu auch eine Anleitung?

Danke schon mal im Voraus.

knutwurst, 2022-03-29:
Hallo @Elektrospeedy ,
danke für die Blumen ;) 

I'll answer in English so that nobody is excluded who might have the same question:

If you take the source code from the current [master](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/archive/refs/heads/master.zip), then so everything is already altered there and you can just compile like that.

If you just want to compile your own code with the error, you can either comment out the line that causes the error for your profile or replace it with my version uploaded here. Here is the example for **MEGA_S_1.4.3:**

https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/0440770618e88d3c24db8f65957c7091b302fa02/ini/i3_mega.ini#L150

Replace the line with:
`TMC26XStepper=https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/8306697/TMC26XStepper.zip`

lg
Olli

Elektrospeedy, 2022-03-29:
Hallo @knutwurst,

ich bedanke mich für die schnelle und hilfreiche Antwort. 
Es hat super geklappt mit dem Kompilieren!

Danke. 👍

ackerthehacker2, 2022-04-07:
The libraries can  be found at:     https://github.com/interactive-matter

I did a replace of trinamic with interactive0matter in platformio.ini