knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #308: [BUG] (short description)

oooBuGooo, 2022-03-20:
Laser reduziert die Geschwindikeit bevor er anfängt zu Feuern

Laser reduces speed before it starts firing

Beim Gravieren von Bilder bremst der Laser vor jedem punkt ab bevor er ihn Lasert, dadurch entstehen tiefere einbennungen.

When engraving images, the laser slows down before each point before lasering it, resulting in deeper levels.

**Expected behavior:** [What you expect to happen]

Sobald ich ein Bild Jarvis modus gravieren möchte stoppt der Laser vor jedem punkt den er lasern möchte  

As soon as I want to engrave an image Jarvi mode, the laser stops before every point it wants to laser

**Actual behavior:** [What actually happens]

Eigentlich sollte der Laserkopf stätig von links nach rechts fahren und den Laser für jenen punkt an und aus schalten ohne das der Laser vor dem Arbeitsschritt abbremst.

Actually, the laser head should move continuously from left to right and switch the laser on and off for that point without the laser slowing down before the work step.

#### Additional Information

https://lightburnsoftware.github.io/NewDocs/Troubleshooting-GRBL.html

Ich habe auf dieser Seite etwas mit Lasermodus gelesen doch weis ich nicht wie ich diesen aktiviere.
muss das in der Firmware eingestellt werden?

I have read something on this page with laser mode but I do not know how to activate it.
does this have to be set in the firmware?

hsd123456789, 2022-04-03:
Hallo, 

darf ich Fragen, wie du den Laser zum laufen bekommen hast in Lightburn?
Hab ihn als Marlin Gerät angelegt, leider wird der Laser nicht befeuert, lediglich die Steuerung funktioniert, und beim "Feuer" Knopf pfeift der Laser nur leise.

Danke dir

oooBuGooo, 2022-04-03:
Du musst den laserbefehl auf m 114 m115 stellen

hsd123456789, 2022-04-03:
Also unter den Marlin Geräteeinstellungen habe ich lediglich die Option Steuerbefehle: M106/M107, M03/M05 und Inline

oooBuGooo, 2022-04-04:
Sorry meinte M106/M107

oooBuGooo, 2022-04-06:
[(2) Common Diode Laser Issues & How To Troubleshoot Them - YouTube](https://www.youtube.com/watch?v=1Kb324MJn-I&t=346s)

Bei 3:55 wird es erklärt doch funktioniert das bei mir so nicht.

At 3:55 it is explained but it does not work for me.

Ich habe jetzt schon öfters gelesen das man in der konsole den laser mode einstellen muss.
Mit $$ komme ich bei mir in der konsole aber nicht weit weil der den befehl nicht erkennt
gibt es noch eine ander möglichkeit den lasermodus zu wechseln?

I have already read several times now that you have to set the laser mode in the console.
With $$ I get with me in the console but not far because the command does not recognize the command.
is there another way to change the laser mode?

closed 2022-05-12 by github-actions[bot]