knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #310: [BUG] (Neopixels install returning an undefined reference error)

TeknikSaiko, 2022-03-23:
### Bug Description

When adding the physical hardware of neopixels and ensuring the hardware works, I tried to fill out the neopixel code in configuration.h and then I built the firmware in platform.io (MEGA_P_DGUS_TMC). The code got built up until the very last part where is resulted in an error of

Archiving .pio\build\MEGA_P_DGUS_TMC\libFrameworkArduino.a
Linking .pio\build\MEGA_P_DGUS_TMC\firmware.elf

****C:\Users\"MY NAME"\AppData\Local\Temp\ccpFS0bN.ltrans2.ltrans.o: In function `GcodeSuite::process_parsed_command(bool)':
<artificial>:(.text+0x5884): undefined reference to `GcodeSuite::M150()'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\MEGA_P_DGUS_TMC\firmware.elf] Error 1****

### Steps to Reproduce

1. Open the folder in VCS and navigate to configuration.h
2. Modify code from lines 3528 to 3538
// Support for Adafruit NeoPixel LED driver
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRBW // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  #define NEOPIXEL_PIN     11       // LED driving pin
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN    5
  #define NEOPIXEL_PIXELS 20       // Number of LEDs in the strip. (Longest strip when NEOPIXEL2_SEPARATE is disabled.)
  #define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 127  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup

3. Go into platformIO --> MEGA_P_DSUS_TMC --> Build

4. Receive error above

**Expected behavior:** [What you expect to happen]

Code finishes compiling successfully into a hex file

**Actual behavior:** [What actually happens]

Failure and Error code 

C:\Users\"MY NAME"\AppData\Local\Temp\ccpFS0bN.ltrans2.ltrans.o: In function `GcodeSuite::process_parsed_command(bool)':
<artificial>:(.text+0x5884): undefined reference to `GcodeSuite::M150()'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\MEGA_P_DGUS_TMC\firmware.elf] Error 1****
[image]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ignotus666, 2022-03-23:
Comment out  `-<src/gcode/feature/leds/M150.cpp>` in ~ line 238 of platformio.ini. I think it should compile then.

TeknikSaiko, 2022-03-23:
> Comment out `-<src/gcode/feature/leds/M150.cpp>` in ~ line 238 of platformio.ini. I think it should compile then.

Thanks! For some reason, I had to edit this out, compile it, and then add in the neopixel code but it worked. I appreciate it!

closed 2022-04-28 by github-actions[bot]