knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #313: [BUG] Anycubic Mega Pro TMC X,Y,Z only at 50% movement

Lunzifer, 2022-03-31:
### Bug Description

**Expected behavior:** 1:1 accurate movement of each axis

**Actual behavior:** After updating the Anycubic Mega Pro with the [MEGA_P_DGUS_TMC_v1.4.3.hex](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/download/1.4.3/MEGA_P_DGUS_TMC_v1.4.3.hex), the each X, Y and Z-Motors only move 50% of what they should. For example: In Cura I want to move X-Asis 10mm but in RL the Movement is 5mm

**Additional Information:** 
- Anycubic Mega Pro (DGUS Display, Trigorilla 0.0.2 Board)
- TMC 2225 Drivers
- Stock Motors

knutwurst, 2022-03-31:
That's not a firmware bug. Your TMC2225 use 1/32 microstepping instead of 1/16. So you need to double the frequency in the firmware or set your drivers to 1/16.

Lunzifer, 2022-03-31:
Thanks for your reply.

I've tried to open the source code according to your manual with VS Code.
After clicking on "Build" i get the following message:

`Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing https://github.com/trinamic/TMC26XStepper/archive/master.zip
PackageException: Got the unrecognized status code '404' when downloaded https://github.com/trinamic/TMC26XStepper/archive/master.zip:
  File "C:\Users\tomis\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 181:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\tomis\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\tomis\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\tomis\.platformio\platforms\atmelavr\builder\main.py", line 162:
    target_elf = env.BuildProgram()
  File "C:\Users\tomis\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\tomis\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\tomis\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\tomis\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 141:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\tomis\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\tomis\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1104:
    project.install_dependencies()
  File "C:\Users\tomis\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 940:
    lm.install(spec)
  File "C:\Users\tomis\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 48:
    pkg = self._install(
  File "C:\Users\tomis\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 93:
    return super(LibraryPackageManager, self)._install(
  File "C:\Users\tomis\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97:
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "C:\Users\tomis\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 137:
    dl_path = self.download(url, checksum, silent=silent)
  File "C:\Users\tomis\.platformio\penv\lib\site-packages\platformio\package\manager\_download.py", line 65:
    fd = FileDownloader(url)
  File "C:\Users\tomis\.platformio\penv\lib\site-packages\platformio\package\download.py", line 39:
    raise PackageException(
====================================================================================================== [FAILED] Took 1.03 seconds ======================================================================================================
Environment    Status    Duration
-------------  --------  ------------
MEGA_P_DGUS    FAILED    00:00:01.028
================================================================================================ 1 failed, 0 succeeded in 00:00:01.028 ================================================================================================ 
Der Terminalprozess "C:\Users\tomis\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'MEGA_P_DGUS'" wurde mit folgendem Exitcode beendet: 1.`

and the error:

`Error: Multiple requests to rebuild the project "Marlin" index have been received! Automatic index rebuilding process has been terminated for 10 minutes. at A.value (c:\Users\tomis\.vscode\extensions\platformio.platformio-ide-2.4.3\node_modules\platformio-node-helpers\dist\index.js:1:790605) at A.value (c:\Users\tomis\.vscode\extensions\platformio.platformio-ide-2.4.3\node_modules\platformio-node-helpers\dist\index.js:1:792714) at A.value (c:\Users\tomis\.vscode\extensions\platformio.platformio-ide-2.4.3\node_modules\platformio-node-helpers\dist\index.js:1:794550) at c:\Users\tomis\.vscode\extensions\platformio.platformio-ide-2.4.3\node_modules\platformio-node-helpers\dist\index.js:1:795...`

knutwurst, 2022-03-31:
Unfortunately this is a known issue. Please see #307

Lunzifer, 2022-03-31:
Ah. Thanks you for the information! The solution has worked now and the click on "Build" succeeded.
But I wasn't able to find the setting where to change the frequency:

> That's not a firmware bug. Your TMC2225 use 1/32 microstepping instead of 1/16. So you need to double the frequency in the firmware or set your drivers to 1/16.

Could you tell me where to find the setting?

Lunzifer, 2022-04-07:
Well. I'm pretty lost now. Checked, edited and tested all options for the MEGA_P_DGUS_TMC where I can edit the microsteps in the firmware but without success :-(

knutwurst, 2022-04-08:
By "That's not a firmware bug." I meant that you cannot change the microstepping within the firmware. You have three options:

1. Take a look at https://wiki.fysetc.com/TMC2225/. There is a table showing the microstepping mode by using the config pins **MS1** and ** MS2**. Yout Trigorilla Board always forces both(!) pins to 5V (vio). Thats perfect for TMC2208 which would use 1/16 microstepping, but as you can see, your TMC2225 use 1/32. You could cut the MS1 leg off which hopefully translates to NC/GND. Otherwise you maxbe need to solder a jumperwire to ground (GND) to ensure, MS1 is on GND and not 5V.
[image]

2. You can use the one-time-programming (OTP) to programmatically configure each stepper driver. This is done outside of the printer and is non reversible. But then, you have 1/16 TMC2225.
See: https://wiki.fysetc.com/TMC2225/#otp-setting-refer-to-tmc2208

3. You can change the steps in the firmware and feed the stepper drivers with double the amount of steps. Just go to this line https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/e3a2aeb4cf9498f51d2fc71d9f8c0d3bf92104c5/Marlin/Configuration.h#L1116 and double every value (e.g. from 80 to 160, from 400 to 800 etc). BUT this utilises the CPU way more than intended and you may get inaccurate prints and also a much lower overall printing speed.

I hope this helps a bit :)

Lunzifer, 2022-04-09:
Hey - many thanks for your mentioned options. First I've tried to jump MS1 to GND, but this caused some kind of short-circuit and the mainboards did not boot.

Then I've removed the MS1 Pin and this has worked perfectly! The motors are so quiet now (max. 10db when moving at max. speed). Maybe you add this detail to your Wiki? If you want, I can provide some pictures and text for this case.

Best regards!