knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #315: [BUG] TMC_DEBUG enabled - Linker error

cpm64, 2022-04-06:
### Bug Description

Mega S 11 TMC BLT, 
Linker error when enabling "#define TMC_DEBUG":

/tmp/ccxB4x4l.ltrans0.ltrans.o: In function `tmc_get_registers(TMC_get_registers_enum, bool, bool, bool, bool) [clone .isra.3]':
<artificial>:(.text+0x223e): undefined reference to `print_hex_long(unsigned long, char)'
/tmp/ccxB4x4l.ltrans0.ltrans.o: In function `GcodeSuite::M122()':
<artificial>:(.text+0x363c): undefined reference to `print_hex_long(unsigned long, char)'
collect2: error: ld returned 1 exit status
*** [.pio/build/MEGA_S_TMC_BLT_11/firmware.elf] Error 1

### Steps to Reproduce

1. Pull the latest from Git
2. Un-comment "#define TMC_DEBUG" in Configuration_adv.h 
3. MEGA_S_TMC_BLT_11 -> Build

**Expected behavior:** [What you expect to happen]
Successful build, firmware binary to be generated

**Actual behavior:** [What actually happens]
/tmp/ccxB4x4l.ltrans0.ltrans.o: In function `tmc_get_registers(TMC_get_registers_enum, bool, bool, bool, bool) [clone .isra.3]':
<artificial>:(.text+0x223e): undefined reference to `print_hex_long(unsigned long, char)'
/tmp/ccxB4x4l.ltrans0.ltrans.o: In function `GcodeSuite::M122()':
<artificial>:(.text+0x363c): undefined reference to `print_hex_long(unsigned long, char)'
collect2: error: ld returned 1 exit status
*** [.pio/build/MEGA_S_TMC_BLT_11/firmware.elf] Error 1
#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

knutwurst, 2022-04-06:
That's not a bug. It can not work because TMC_DEBUG requires the stepper drivers to have a communication port like UART or SPI to work.

cpm64, 2022-04-06:
Can be true, however this debug enabled worked in rev 1.3.1 for example.

closed 2022-05-12 by github-actions[bot]