knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #322: Wifi mit dem Anycubic mega S

bodo328, 2022-05-04:
Hallo

Ich habe meinem Anycubic Mega S nach Anleitung versucht WIFI fähig zu machen habe die Firmware ESP33D installiert und die Index.html.gz Datei aufgespielt und habe zugriff auf den ESP01. Das funktioniert alles einwandfrei. 

Nun meine Frage warum kann ich den Drucker damit nicht steuern bzw. wenn er arbeitet zeigt er mir kein Gcode oder Temperatur an auch auf den Inhalt der SD Karte habe ich keinen Zugriff. 
Welche Einstellung habe ich übersehen? habe nach Anleitung gearbeitet, Lötverbindung habe ich auch geprüft und mit dem Ohmmesser durchgeklingelt anderen ESP genommen ältere Software eingespielt es geht einfach nix

Wer kann helfen oder hat noch eine Idee

mit freundlichen Grüßen bodo328

tahir51, 2022-05-04:
Hey Bodo, welche Firmware hast du auf deinem Drucker draufgespielt? Soweit ich das herauslesen konnte, wurde die Wifi-Funktionalität ab V1.3.0 abgeschaltet. (Außer du hast diese in der configuration.h datei aktiviert)

bodo328, 2022-05-05:
Da habe ich die 1.4.0 und die 1.4.3 ausprobiert werde dann mal die 1.3.0 probieren.
Ansonsten muss ich sehen wie ich das mit Platform io selbst compiliert. Habe mir das zwar gestern angeschaut aber habe es nicht hinbekommen . 
Vielen Dank thair51 für den Tipp werde heute Nachmittag  mal die Version 1.3.0 auf den Drucker aufspielen

bodo328, 2022-05-05:
Vielen Dank 
mit der Version 1.3.0 Funktioniert alles

bodo328, 2022-05-05:
Vielen DANK

closed 2022-05-05 by bodo328

RasputinBerry, 2022-06-13:
Ich hatte die 1.4.4. geladen um dann nach Nicht-Funktion der WLAN Anbindung diesen Beitrag zu finden (In der Anleitung zum WLAN Einbau steht AB  und nicht BIS 1.3.0 ... ).
Jetzt habe ich die 1.3.0 aufgespielt und es geht leider noch immer nicht. 
Die Hardware des ESP-01 funktioniert am Programmer Stand-alone ohne Probleme, Verkabelung am Mainboard mehrfach kontrolliert. Ich finde keinen Fehler (mehr) ...

Hat jemand noch eine Idee ?! Danke.