knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #324: [BUG] Firmware ignoring FAST_PWM_FAN / FAST_PWM_FAN_FREQUENCY

Unkn0wn-MDCLXIV, 2022-05-06:
### Bug Description

My 5015 blower fan makes terrible high pitch noises when controlled by PWM. I enabled `FAST_PWM_FAN ` and tried different `FAST_PWM_FAN_FREQUENCY` values, but the frequency does not change at all. The noise stays completely the same.

### Steps to Reproduce

1. Set` #define FAST_PWM_FAN`
2. Set `#define FAST_PWM_FAN_FREQUENCY ` to any value just to test

**Expected behavior:**
The noise of the pitch should change at least.

**Actual behavior:**
Everything stays the same.

#### Additional Information
`FAN_SOFT_PWM` seems to work just fine, but I rather want to use hardware PWM.

knutwurst, 2022-05-06:
You only have two options. Either you use hardware PWM OR software PWM with fast timings. FAST_PWM only refers to the SoftPWM Values. That's what it says in the comments.

Unkn0wn-MDCLXIV, 2022-05-06:
I did not use Software PWM in combination with fast pwm.

knutwurst, 2022-05-06:
Yes, I know - but fast pwm relies of software pwm :) 
There is no "fast hardware pwm".

Unkn0wn-MDCLXIV, 2022-05-06:
Oh, my bad then.
Is there no way to control the hardware pwm to get rid of the noises?

knutwurst, 2022-05-06:
Unfortunately, no. But what speaks against software PWM? The PWM cycles are only generated during computing breaks anyway, so you don't get any performance problems.

The only thing to note, however, is that if you switch to SoftwarePWM, the heaters are also controlled via SoftwarePWM. If your PWM frequency is too low or the PWM cycles are too long, it could kill the MOSFETs of the heatbed. It hasn't happened to me yet, but theoretically it's possible. For this reason I also use HardwarePWM so that the PID runs correctly.

Unkn0wn-MDCLXIV, 2022-05-06:
So you use both?

the problem with software pwm is that I can hear the fan spinning up and down.
It’s not as annoying as the high pitch noise tho.

closed 2022-06-11 by github-actions[bot]