knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #325: [FR] M808

GITdan80, 2022-05-08:
### Description

Bitte M808 Repeat-Funktion hinzufügen

### Feature Workflow

Laut Marlin Wiki ist in Marlin 2.0.8 die Repeat-Funktion M808 enthalten.
Zum Einlaufen von neuen Achsen oder um zu testen, ob die Motoren oder Stepper Treiber ein Temperaturproblem haben, wäre diese Funktion sehr nützlich, da man sehr einfach endlose Fahrweg-Schleifen programmieren könnte.

stklcode, 2022-05-08:
Ist laut Konfiguration nicht ausgeschlossen. Funktioniert der Befehl nicht?

https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/e3a2aeb4cf9498f51d2fc71d9f8c0d3bf92104c5/platformio.ini#L282

GITdan80, 2022-05-09:
Hallo,

Ja, ich habs mehrfach ausprobiert und M808 wird jedes Mal ignoriert.
Das Flashen deiner Firmware hat aber auf jeden Fall funktioniert. Danach lade ich auch immer die Firmware  defaults,  speichere und starte den Drucker neu.
Wenn ich eine Schleife mit 100 Wiederholungen schreibe, fährt der Drucker nur 1x hin und her und bleibt anschließend stehen.

LG, Daniel

PR #326 (2022-05-10): config: enable g-code M808 for repeat markers (#325)
### Description

Enable G-Code [M808](https://marlinfw.org/docs/gcode/M808.html) - Repeat Marker

### Benefits

Repeat markers and loops are supported

### Configurations

part of code changes

### Related Issues

resolves #325
closed 2022-05-10 by knutwurst