knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #327: [BUG] (short description)

rstalde, 2022-05-10:
Have you read all Wiki articles in cluding the FAQ? Yes
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki

A little problem (or a late Easter egg) with an Anycubic I3 Mega S and your current firmware version:
I tried to print a part for an Ender printer.
The filename started with "ender", The print should be done from the SD card.
Problem: The filename just doesn't show up in the printer's selection menu. All *.gcode files that start with "ender" are invisible ... 
Only after renaming the file to "11ender.gcode" only, printing was possible.
Is this a bug, an undocumented option or I'm too stupid? ;)

Greetings from the river Rhine
Ralf

Translated with www.DeepL.com/Translator (free version)

stklcode, 2022-05-17:
Interesting thing.

Just tested myself, confirmed. Apparently it's not "ender", but the keyword "end". "end.gcode", "end123.gcode", etc. all cause the same issue.

FIles alphabetically before "end" are still shown, enything afterwards is not. Only the TFT outpus is affected, Octoprint shows the files on SD card.

knutwurst, 2022-06-17:
But that only applies to the blue-yellow dgus clone tft, right?

stklcode, 2022-06-17:
My findings are from a DGUS-clone, right (don’t have a different one to test)

There should be some more or less trivial solutions, if it’s a TFT bug (rewrite the filename, add some prefix, or similar), but I think we should check the original FW/sources first for any hints on that.

rstalde, 2022-06-17:
I have the problem with the white / blue display, the old version of the Anycubic Mega. The effect also occurs with the firmware 1.4.4 on the motherboard.

closed 2022-07-24 by github-actions[bot]