knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #328: [BUG] Mega_X_v1.4.3 Display zeigt Anycubic logo und friert ein

Catsuoo, 2022-05-11:
### Bug Description
Hallo, bei der FW v1.4.3 zeigt mein Mega X (erste Generation mit DWIN/DGUS II Display) nach dem Startvorgang nur noch das Anycubic Logo an und reagiert auf keine Eingaben über den Touchscreen. Der Zugriff über Pronterface/Cura funktioniert.

### Steps to Reproduce
1. Firmware vor dem Update MEGA_X_v1.3.1.hex
2. Update der Firmware auf MEGA_X_v1.4.3.hex via Cura v4.13.1
3. Neustart des Druckers
4. Startanimation wird angezeigt
5. Menü wird für wenige Sekunden angezeigt (Fehlermeldung, dass der Filamentsensor nicht angeschlossen ist. Das ist auch richtig so)
6. Anycubic logo wird angezeigt und geht nicht mehr weg
 [image]
7. Touchscreen reagiert nicht
8. Update der Firmware zurück auf MEGA_X_v1.3.1.hex via Cura v4.13.1
9. Drucker arbeitet normal
10. siehe Punkt 1.

#### Additional Information
Vorher installierte FW Versionen bei denen dieses Problem nicht aufgetreten ist:
MEGA_X_v1.1.7.hex
MEGA_X_v1.2.0.hex
MEGA_X_v1.3.1.hex

knutwurst, 2022-06-11:
I'm trying to reproduce the bug. Is there an SD card in the printer? If not, does that change anything?

Catsuoo, 2022-06-11:
Hi, thank you for your assistance! I would guess that the SD card was inside, but I'm not sure. I will retry with and without SD and let you know if it makes a difference.

Catsuoo, 2022-06-11:
It doesn't matter if the SD card is inserted or not 

The issues is the same for the new version MEGA_X_v1.4.4.hex too.

e0temp, 2022-06-12:
I can confirm the described problem. The state of the sd card does not matter. The problem can be avoided, when putting in filament into the filament sensor at start-up.

Catsuoo, 2022-06-12:
Thank you for the info. 

I do not use the sensor because there is not enough space for it in my cabinet. But i will try to connect it with a piece of filament inside to bypass the issue.

closed 2022-07-18 by github-actions[bot]