knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #331: [FR] (feature request title)

OHacks, 2022-05-21:
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.0 Transitional//EN">
<html>
<head>
	<meta name="generator" content="LibreOffice 6.4.7.2 (Linux)"/>
</head>
<body lang="de-DE" link="#000080" vlink="#800000" dir="ltr">

<H1>Messung Vref/Imax für TMC2208 Anycubic Mega X:</H1>

Ich habe seit kurzem einen Anycubic Mega X und habe diesen nach Deiner Anleitung auf die Treiber TMC2208 umgerüstet. Da ich bisher nur Vermutungen über den einzustellenden Maximalstrom für die TMC2208 gefunden habe, habe ich die verwendeten Referenzspannungen der original verbauten A4988 gemessen. Anschließend habe ich von einem den Kühlkörper entfernt, um die Werte der Shuntwiderstände ablesen zu können:

Verbaut sind 100 mOhm (R100) Shuntwiderstände  als Rsense (siehe Bild am Ende). Damit ergibt sich folgender Maximalstrom in Abhängigkeit der Referenzspannung:
Imax = Vref / (8 * Rsense), mit Rsense = 0.1 Ohm
Imax = Vref / (8 * 0.1 Ohm) = Vref * 1.25 / Ohm

Hier die Tabelle mit den Meßwerten (A4988 Vref) und sich den daraus ergebenen Maximalströmen und Referenzspannungen für die TMC2208:

Außenseite  |||||| zur Gehäusemitte
-- | -- | -- | -- | -- | -- | --
  |   | X | Y | Z | E0 | E1=Z2
A4988 | Vref 		 | 0,905 V | 0,978 V| 0,916 V| 0,749 V| 0,922 V
 Maximalstrom | Imax 	 | 1,13 A| 1,22 A| 1,15 A| 0,94 A| 1,15 A
TMC2208 | Vref 		 | 1,13 V | 1,22 V| 1,15 V| 0,94 V| 1,15 V

(Die Messung erfolgte genau so, wie die Messung bei der Einstellung der Referenzspannung beim TMC2208!)

Ich habe die Werte bei mir entsprechend bei den TMC2208 eingestellt, der Drucker funktioniert damit einwandfrei!
Für die TMC2208 lassen sich gut Kühlkörper mit 9mm x 9mm x 5mm (5mm = Höhe) verwenden.

[image]

Du darfst diese Informationen und das Bild gerne im Wiki verwenden!

Vielen Dank für deine Arbeit und die Bereitstellung der diversen compilierten Firmware Varianten!

Gruß
Oli

</body>
</html>

knutwurst, 2022-05-29:
Hallo Oli, hier Olli :D
Vielen Dank für die Doku! 

Ich werde dies bei Zeiten mit in die offizielle Doku übernehmen. Kurz dachte ich, du hättest einen Denkfehler, da ich mich bei den TMC Treibern meistens auf den RMS des Stromes konzentriere. Aber ja, dein eingestellter Maximalstrom stimmt natürlich.

Ich weiß nur nicht, inwieweit man Anycubic mit den default-Werten trauen kann, da diese vermutlich auch nur grob eingestellt wurden. 

Ein guter Anhaltspunkt ist immer dieser Rechner hier: https://learn.watterott.com/de/silentstepstick/faq/

Trotzdem ist deine Erkenntnis natürlich Gold wert, wenn man dies auf TMC2208 bezieht.

Gruß
Olli

OHacks, 2022-05-29:
Hallo Olli,wenn Du magst, könnten wir andere Nutzer bitten, die von mir beschriebene Messung ebenfalls durchzuführen und die Werte zu übermitteln (die müssen ja nicht den Kühlkörper runter reißen), dann könnte man einen Mittelwert bestimmen, der in etwa dem Zielwert von Anycubic entsprechen sollte. 10 Messwert-Sätze sollten schon belastbare Mittelwerte erzeugen.Den relativ kleinen Strom für den Extruder fand ich überraschend, scheint aber in Ordnung zu sein!GrußOli Von meinem/meiner Galaxy gesendet

chilippso, 2022-06-08:
> Den relativ kleinen Strom für den Extruder fand ich überraschend, scheint aber in Ordnung zu sein!

Das wird schätzungsweise an der kurzen Übersetzung im Extruder liegen.

knutwurst, 2022-06-09:
> Das wird schätzungsweise an der kurzen Übersetzung im Extruder liegen.

Richtig. Die weit verbreitete Annahme, dass der E0 Treiber eine deutlich höhere Vref benötigt wurde schon mehrfach widerlegt. Das ist wohl dem ersten i3 Mega geschuldet, der noch mit einem MK8 Extruder (Ritzel direkt auf der Motorachse) ausgeliefert wurde. Der Titan (clone) hat einen viel kleineren und schnelleren Motor verbaut, der einen geringeren Strom benötigt.

chilippso, 2022-06-12:
> Der Titan (clone) hat einen viel kleineren und schnelleren Motor verbaut, der einen geringeren Strom benötigt.

Kurze Story hierzu: 1,2V sind zu viel für E0. Hatte beim letzten mal nicht aufgepasst und es dann einfach testweise mal so gelassen. Der Stepper wird viel zu heiß und ich hatte Under-Extrusion Probleme. Habe E0 jetzt wieder auf 1,0V.

phlipse, 2022-06-27:
How did you measure the Vref exactly? Which test points did you use? Then I can provide the measurements from my Mega X.

OHacks, 2022-06-27:
You have to measure the voltage between GND and the potentiometer of the original driver (A4899) (take a look at the attached images)

Take care that:

- the power of the 3d printer has to be turned on!
- you do not change the potentiometer setting!
- make sure your voltmeter is in voltage mode!
- make sure you do not create any shortcuts!
- measure it for all 5 drivers in note the position: X   Y   Z   E0   E1=Z2 
- X driver is positioned to the outside of the printer
- E1=Z2 driver is positioned to the inside of the printer

[image]

[image]

shakin89, 2022-07-12:
Can i use this vref for a mega S?

shakin89, 2022-07-13:
well, i could measure the original voltage, cause i still have the a4988 from anycubic. but i've got them off the board, so i don't really know which one was for x, y or z.... 
by the way, i'll mount them again on the board, will measure the voltage trying to figure which one was for x, y or z and post the values here.

rstalde, 2022-07-18:
@shakin89
The order of the stepper drivers comes from the opening post.
"Aussenseite" means the side facing the housing plate.
"zur Gehäusemitte" means the side that points to the power supply unit.

chilippso, 2022-07-18:
> The order of the stepper drivers comes from the opening post.

I guess the problem is to recall which driver belonged to which position and not how they are ordered on board / in the table

I also have my old drivers somewhere and I will also measure their original vrefs. But I also can not tell anymore which driver belongs to which particular motor / axis.

EasyG0ing1, 2022-07-23:
Here are my stock Mega X driver voltage readings:

Board: TriGorilla V0.0.2
```
E1 - 0.900
E0 - 0.780
 Z - 0.907
 Y - 1.006
 X - 0.917
```
Readings were taken with a Fluke 17B+ Meter. And all I did was flip the printer, remove the bottom, plugged it in, powered it on and took the readings (after dismounting the cooling fan above the drivers).

EasyG0ing1, 2022-07-23:
If anyone is interested, they can use [this link](https://simtechdata.com/megax/) to record their voltage readings, then when they click submit, it will kick out the text that they can then copy and paste into a response into this issue and all of the responses will be formatted into an HTML table for uniformity.

EasyG0ing1, 2022-07-27:
@OHacks By the way, using the same calculations as you used to set the voltage of the 2208 drivers worked out perfectly for me, except the extruder motor on the Mega X makes a "clunking" sound if there is any tension between the filament feed spool and the extruder motor. When I spin the filament spool so that there is no tension, then it doesn't make that noise, so I'll have to increase the voltage on that driver only, but the rest are working perfectly.

gabber130779, 2022-07-29:
Printer: Mega X

Board: V0.0.2

<table style="text-align:center">
  <tr>
    <th>Board Label</th>
    <th>Voltage</th>
  </tr>
  <tr>
    <td>E1</td>
    <td>1.15</td>
  </tr>
  <tr>
    <td>E0</td>
    <td>1.10</td>
  </tr>
  <tr>
    <td>Z</td>
    <td>1.15</td>
  </tr>
  <tr>
    <td>Y</td>
    <td>1.19</td>
  </tr>
  <tr>
    <td>X</td>
    <td>1.13</td>
  </tr>
</table>

phlipse, 2022-07-31:
Printer: Mega X

Board: TriGorilla V0.0.2

<table>
  <tr>
    <th>Board Label</th>
    <th>Voltage</th>
  </tr>
  <tr>
    <td>E1</td>
    <td>0.9322</td>
  </tr>
  <tr>
    <td>E0</td>
    <td>0.7835</td>
  </tr>
  <tr>
    <td>Z</td>
    <td>0.9296</td>
  </tr>
  <tr>
    <td>Y</td>
    <td>0.9790</td>
  </tr>
  <tr>
    <td>X</td>
    <td>0.9314</td>
  </tr>
</table>

DMM: Brymen BM869s

OHacks, 2022-08-01:
> Does anyone use the TMC2208 from Watterott? The chip + potentiometer is on the downside of the PCB. How did you basically adjust the voltage and where did you put the heat sink?

The TMC2208 shall be cooled through the PCB (not directly on the top of the chip). There are several connections (vias) on the PCB where you should put the heat sink. Make sure that the heat sink does not contact any pins or elements!!!

There is a hole over the potentiometer, you can adjust and measure the voltage through the hole. 
Best: measure the voltage while adjusting the potentiometer by connecting the voltage meter with a clip to the screw driver.

OHacks, 2022-08-01:
@gabber13077

<p dir="auto">Printer: Mega X</p>
<p dir="auto">Board: V0.0.2</p>

>Board Label | Voltage
>-- | --
>E1 | 1.15
>E0 | 1.10
>Z | 1.15
>Y | 1.19
>X | 1.13

When these are the measured voltages of the A4988 then your A4988 seem to have a different shunt resistors, so you can not use the calculations here. If you want to be sure you can remove one heat sink and write which values are noted on the two resistors shown in the image of my first post.

JernejL, 2022-08-01:
I know you are talking in german, but i can add some information here. 
 
I print A LOT of petg on i3 mega x and use tmc2209, but this could affect to 2208 as well. 
 
For extruder, voltage depends on your extruder. 
1 volt is enough if you use dual gear extruder, otherwise you need higher voltages. 
 
This is my setup that i use: 
 
X 0.95
Y 1.0v (or else i get layer shifts with heavier or higher models)
Z 0.95
E0 1.1v for titan extruder, dual gear extruder with this is loud -> 1.0v is ok for dual gear extruders
E1 ( Z2 ) 0.95v

OHacks, 2022-08-01:
I create an average of the Vref values from phlipse and EasyG0ing1 (and mine), the maximum current and the Vref for the TMC2208 to initial post. As you can see the values are very similar.

JernejL, 2022-08-01:
@OHacks ofcourse, that is not a problem, your values are ok. 
I'm just adding information on what worked for me by experimenting, especially info that if you replace or upgrade to dual gear extruder, you can reduce extruder stepper driver voltage.

elodur, 2022-08-07:
Printer: i3 Mega S 

Board: Trigorilla V0.0.2

<table>
  <tr>
    <th>Board Label</th>
    <th>Voltage</th>
  </tr>
  <tr>
    <td>E1</td>
    <td>0.936</td>
  </tr>
  <tr>
    <td>E0</td>
    <td>0.787</td>
  </tr>
  <tr>
    <td>Z</td>
    <td>0.947</td>
  </tr>
  <tr>
    <td>Y</td>
    <td>1.014</td>
  </tr>
  <tr>
    <td>X</td>
    <td>0.908</td>
  </tr>
</table>

DMM: Uni-T UT61E

EasyG0ing1, 2022-08-08:
As a point of interest perhaps for some people, I replaced my drivers with [these 2209's](https://www.amazon.com/dp/B09LCZJV1F), and instead of using the double sided tape that the heat sinks come with, I scraped off the tape and used heat sink epoxy for a better bond to the board. However, I didn't think to check the height of the heat sinks, and it turns out that they are so tall that the original placement of the cooling fan was no longer possible. 

I bent the bracket to get more height out of it, but the bracket was so brittle that the bend ended up snapping off.

I designed a bracket, which can be [downloaded here](https://www.thingiverse.com/thing:5451670), or if you would like to edit it, you can grab the [tinkercad version](https://www.tinkercad.com/things/0M1D73msxN9) here. You can use the original screws with the bracket, they will cut threads into the plastic without any issues at all.

I ended up replacing the single cooling fan with two different sized fans - a [40mm](https://www.amazon.com/dp/B00ZB7TD4G) and a [50mm](https://www.amazon.com/dp/B0083A0CBG) (they are whisper quiet) which provides complete coverage of the heat sinks. However, even with the new fans literally resting against the new heat sinks, the exposed side of the fan blades on the 50mm ended up resting against the bottom cover plate which prevented it from spinning, so I had to use a Dremel to cut out part of the cover plate so the fan can spin. The feet on the corners of the printer provide plenty of clearance so that air flow isn't a problem.
[image]

chilippso, 2022-08-08:
> However, I didn't think to check the height of the heat sinks, and it turns out that they are so tall that the original placement of the cooling fan was no longer possible.

I had the same issue and I also didn't want to use the tape but a heat sink adhesive. So I just removed the heat sinks on the original driver and stick them to the 2209s. No height issues.

EasyG0ing1, 2022-08-08:
> I had the same issue and I also didn't want to use the tape but a heat sink adhesive. So I just removed the heat sinks on the original driver and stick them to the 2209s. No height issues.

I wish I had thought of actually checking the size of the heat sinks before using the epoxy to adhere them. And since the entire surface area of the heat sync makes full contact with the PCB, I didn't want to risk removing them as I could have possibly damaged the drivers. Hopefully, our comments can save someone else from this "gotcha".

If anyone has any suggestions for replacing the part cooling fan with something quieter, I'd love to see them.

elodur, 2022-08-08:
Take some sanding paper and reduce the height of the heatsinks.

rstalde, 2022-08-13:
For another project (Ender 3) I had to calculate the motor current Imax from Vref ...
The formula from the documentation of the drivers TMC2208 and 2225 fits also here ;)
So there is the Excel / OpenOffice file attached, which calculates the current from Vref and Rsense.
Have fun using and redistribute.

[TMC2208_Motorstrom.xlsx](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/9332052/TMC2208_Motorstrom.xlsx)

shakin89, 2022-09-15:
Sorry for the late reply, i've measured the a4988 vref on 

Printer: i3 Mega S

Board: Trigorilla V0.0.2

Since i've substituted them with tmc2208, i don't know in which order they were. 
i think this is the corerect order

|Board Label|Voltage|
|----|-------|
| E0 | 0,783 |
| E1 | 0,782 |
| X  | 0,817 |
| Y  | 0,931 |
| Z  | 0,969 |

EasyG0ing1, 2022-09-16:
@shakin89 
> Since i've substituted them with tmc2208, i don't know in which order they were. i think this is the corerect order

Here are the drivers labeled to match the table 

[image]

shakin89, 2022-09-16:
The problem is to recall which driver belonged to which position and not how they are ordered on board.

I had my old drivers in a box, and i have measured their original vrefs. As i sad before, i've replaced a4988 with tmc2208 months ago, and I cannot tell anymore which driver belongs to which particular motor / axis.
I hope my measurement can be useful.

knutwurst, 2022-09-16:
@shakin89 ,
Take a look at the first post. The order is shown there.

The Trigorilla board has 2 Z slots, but they work on the same channel. That's why the i3 Mega uses X and Y als normal X ynd Y ports and Z belongs to the first Z axis. E1 controls the second Z axis.

E0 is the extruder.

EasyG0ing1, 2022-09-16:
> The problem is to recall which driver belonged to which position and not how they are ordered on board.
>
> I had my old drivers in a box, and i have measured their original vrefs. As i sad before, [...]

If you're really wanting to get the true values, you could plug them back in and look at the values posted by others and if your drivers are close to those numbers then you could probably reliably pick the one closest to a given number and assume it was for that stepper ... I got lucky, my driver boards were still connected at the perforation marks so they only fit back in one way which was convenient when I wanted to go back and the get the values.

leitenky, 2022-10-05:
What is the maximum vref that can be used on the Mega X with 2208's? Currently If I am much below 1.4 I get all kinds of small layer shifting. My print speeds are 60mm/s and travel is under 100mm/s in Cura. Belts, pullies and V wheels have all been checked as well. Should I look at going to 2209's as I've really struggled to get the Y axis reliable

knutwurst, 2022-10-05:
1.4V ist way too high. 
I would recommend staying between 1.0 and 1.3V. Your layershift is certainly due to the fact that the drivers overheat and therefore switch off for a short time.

leitenky, 2022-10-05:
Yeah I started at 0.95 and kept turning up as I kept having the issue.  I've tried 2 different 2208 drivers eryone and BBT and both end up around 1.4v if I don't want all the missed steppes.  It is usually only a couple mm  per layer but anything large and the part usually ends up in the trash

leitenky, 2022-10-05:
Ok I'll check them again I guess.  I did put a larger noctura cooling fan so and I know it's functioning

closed 2022-11-11 by github-actions[bot]

toopaq, 2022-12-11:
Printer: Anycubic i3 Mega

Board: Trigorilla 1.0

<table>
  <tr>
    <th>Board Label</th>
    <th>Voltage</th>
  </tr>
  <tr>
    <td>E1</td>
    <td>0.94</td>
  </tr>
  <tr>
    <td>E0</td>
    <td>1.03</td>
  </tr>
  <tr>
    <td>Z</td>
    <td>0.94</td>
  </tr>
  <tr>
    <td>Y</td>
    <td>1</td>
  </tr>
  <tr>
    <td>X</td>
    <td>0.85</td>
  </tr>
</table>

DMM: McVoice M-345pro

I have taken over the values directly from A4988 into the new TCM2208 drivers without any adjustment (before reding all this) and printed now for about 15h with the new drivers without any issues. I hope undervoltage will not have any side effects.

Boostie76, 2023-01-04:
Printer: Anycubic i3 Mega S

Board: Trigorilla v0.02

<table>
  <tr>
    <th>Board Label</th>
    <th>Voltage [V]</th>
  </tr>
  <tr>
    <td>E1</td>
    <td>0.907</td>
  </tr>
  <tr>
    <td>E0</td>
    <td>0.784</td>
  </tr>
  <tr>
    <td>Z</td>
    <td>0.923</td>
  </tr>
  <tr>
    <td>Y</td>
    <td>1.016</td>
  </tr>
  <tr>
    <td>X</td>
    <td>0.926</td>
  </tr>
</table>

leonel85, 2023-01-21:
Printer: Mega-X

Board: Trigorilla 1.1

<table>
  <tr>
    <th>Board Label</th>
    <th>Voltage</th>
  </tr>
  <tr>
    <td>E1</td>
    <td>0.900</td>
  </tr>
  <tr>
    <td>E0</td>
    <td>0.770</td>
  </tr>
  <tr>
    <td>Z</td>
    <td>0.910</td>
  </tr>
  <tr>
    <td>Y</td>
    <td>0.945</td>
  </tr>
  <tr>
    <td>X</td>
    <td>0.910</td>
  </tr>
</table>