knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #334: Anycubic Mega X - Druckbereich anpassen wegen direkt Extruder

AloisE30, 2022-05-30:
Hallo!

Zuerst einmal vielen Dank an die tolle Arbeit von @knutwurst.

Leider funktioniert der fertige MEGA_X_TMC_BLT_11_v1.4.4.hex bei meinem Drucker nicht perfekt, da mein X-Verfahrweg durch den Einbau eines Sprite Extruders um ca. 30mm verkürzt ist.
Ist es möglich die Arduino-Files zu den vorgefertigten *.HEX zu verlinken damit der User ggf. noch kleinigkeiten ändern kann?

Vielen Dank

knutwurst, 2022-05-30:
Schau mal bitte ins [Wiki](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki). Dort ist eine genaue Anleitung, um die Firmware selbst zu kompilieren inkl. aller Änderungen.

AloisE30, 2022-05-31:
Guten Morgen!
Die Anpassung der Druckbettgröße über den Slicer hilft in diesem Falle nicht, da das Bed-Leveling (G29) bereits scheitert.
Der Druckkopf fährt in X-Richtung mechanisch an den Anschlag und der Riemen springt über.
Danach geht natürlich alles in die Hose...

knutwurst, 2022-05-31:
Ja, korrekt. Aber darauf wollte ich auch nicht hinaus. Du musst dir Firmware selbst kompilieren und das Druckbett dort anpassen.

Die Codeänderungen befinden sich hier:
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/41d069464e3f762c93502bec04245dc0866a19e8/Marlin/Configuration.h#L1925

Und auch wenn ich schon auf das Wiki hingewiesen habe, hier noch mal der Deeplink zum Kompilieren: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Howto:-Firmware-mit-PlatformIO-selbst-kompilieren

AloisE30, 2022-05-31:
Vielen Dank für den Support, ich versuch mal mein Glück.
Bei meinem Tevo Tarantula (Modell Frankenstein) hab ich fast eine Woche gebraucht, bis die Marlin 1.1.9 so lief.

closed 2022-07-07 by github-actions[bot]