knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #335: [BUG] Any_mega_x - Z achse nach Firmware update nur noch 50%

SmaugGold, 2022-05-30:
Hallo Zusammen.

Vorab schon mal danke bei der Lösungsfindung.

Es geht um einen Anycubic mega x. Ein ähnliches Verhalten wie bei Artikel #https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/313 . Bei mir betriff es aber nur die Z Achse und es sind TMC2208 verbaut.

### Bug Description

Nach Anbau eines original BLTouch und einer Firmware update von Marlin 2.0.5.4 auf Marlin 2.0.9.2 (Marlin-2.0-x-Anycubic-i3-Mega-S-master)  fährt die Z Achse nur noch 50%.

### Steps to Reproduce

1. Start des Drucker und Homing ohne Probleme. 
2. Auto Bed Leveling über den BLtouch ohne Probleme möglich.
3. Druck Start von Cura über Octoprint zum Drucker ohne Probleme möglich. Allerdings fährt der Drucker zu wenig in der Z höhe. 
    Manueller Abbruch des Druckauftrags.
5. Z Achse manuell, über das Menü des Drucker , +10mm. Achse fährt aber nur +5mm
6. Gleiches Verhalten über Octoprint oder Cura.

#### Additional Information

BLTocuh:
Die Strom Versorgung ist über S1 und + / -  sind getauscht.  Zmin (weisser Draht) ist an IC2 SCL angeschlossen. In der Firmware ist der  Pin angepasst (Z_MIN_PROBE_PIN 21) Der BlTouch funktioniert ohne Probleme.

TMC 
Es sind alle durch TMC2208 ersetzt. Hatte noch nie Problem mit denen. Bei der Firmware wurde die Library ersetzt wie hier beschrieben:
#https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/307

Firmware
Firmware wird über Visual code und PlatformIO Manuell durch Project Tasks (MEGA_X_TMC_BLT11) Aktualisiert.
Vor der Firmware Aktualisierung Stimmte alles. Die Aktualisierung wurde durchgeführt, da die Alte Version wohl einen Fehler mit dem BLTouch hatte.
Ändern der Z Stepps von 400 auf 800 Funktioniert- > Da das allerdings heissen würde Reduzierung der Druck Geschwindigkeit und mehr Belastung für den CPU, kann das keine Lösung sein.

Meine IT Kenntnisse sind nur rudimentär. Bin mir also nicht ganz sicher ob ich einfach was übersehen habe.

vielen Dank für eure Hilfe.

knutwurst, 2022-05-30:
Hallo @SmaugGold ,
das Verhalten ist völlig normal, wenn du zwischenzeitlich mal eine falsche Firmware mit nur 400 Steps der Z Achse geflasht hast und diese Werte nun im EEPROM stehen. 

Der Mega X benötigt in der Tat 800 Z Steps pro Millimeter (und nicht nur 400 wie der Mega M/S/P). 

Zum Vergleich - MEGA S:
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/41d069464e3f762c93502bec04245dc0866a19e8/Marlin/Configuration.h#L1108

Und hier - MEGA X:
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/41d069464e3f762c93502bec04245dc0866a19e8/Marlin/Configuration.h#L1112

Wenn die neu geflashte Mega X Firmware nur 400 Steps auf der Z Achse hat, so wurden die Defaults nicht geladen und somit keine der internen Werte geladen, sondern die, die schon im EEPROM standen. 

Ich hoffe das hilft dir :) 

lg
Olli

SmaugGold, 2022-05-30:
Hallo Knutwurst.

Vielen dank für die rasche Antwort.

Mhm dann muss ich wohl beim Extruder Kalibrieren denn falschen wert geändert haben.

Ich werde ihn wider auf 800 stellen. 
Danke für den Support und die Tolle Arbeit. Ein Kaffee ist unterwegs ;)

closed 2022-05-30 by SmaugGold