knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #336: [Documentation] Document or Add Mega P pin swap to Firmware

Bat-Shark, 2022-05-31:
Hoping to get some clarification on the pin swapping if following the below guide. Swapping pin 2 to pin 3 does not do what I expected. Using the X Stop header added to the board per the Mega P documentation. 

```
#ifndef Z_MIN_PROBE_PIN
  #define Z_MIN_PROBE_PIN                      3    // PATCH: Knutwurst
#endif
```

https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/210#

Bat-Shark, 2022-06-01:
I originally did just set it to 3, and not swap. I have however just tried swapping correctly and still no luck. It starts the auto bed leveling and just goes straight to X min. I made the changes in the pins_RAMPS.h as indicated, recompiled and flashed both _10 and _11 with no luck. In theory this should be the MEGA_P_DGUS_BLT_10 firmware. I've double checked the wiring for both the 2 pin connector on the X-STOP and the servo underneath, having swapped the 5v and ground, it seems to be correctly oriented and plugged into the only header. Which is why I was thinking it was the pin swap that was the issue.

Bat-Shark, 2022-06-01:
Wondering if it's a conflict with lines 405 in pins_postprocess.h

```
 #ifdef X_STOP_PIN
  #if X_HOME_TO_MIN
    #define X_MIN_PIN X_STOP_PIN
    #ifndef X_MAX_PIN
      #define X_MAX_PIN -1
    #endif
  #else
    #define X_MAX_PIN X_STOP_PIN
    #ifndef X_MIN_PIN
      #define X_MIN_PIN -1
    #endif
  #endif
#elif X_HOME_TO_MIN
  #define X_STOP_PIN X_MIN_PIN
#else
  #define X_STOP_PIN X_MAX_PIN
#endif
```

Bat-Shark, 2022-06-06:
Update for no one in particular. Swapped the A5 and B5 in the connector. Now I get continuity between the X stop on the frame and the A6 pin on the hotend pcb. Should we also disconnect the X Stop???

Bat-Shark, 2022-06-06:
Back on topic slightly, but the documentation could also include the fact that the mega pro homing on the X side requires the fan brackets to be attached, otherwise the sled will not actually hit the X min

knutwurst, 2022-06-17:
> Should we also disconnect the X Stop???
No you don't need to. It's fine to just swap the cables or the pins via firmware modification.

> Back on topic slightly, but the documentation could also include the fact that the mega pro homing on the X side requires the fan brackets to be attached, otherwise the sled will not actually hit the X min
I think this is not necessary. I mean.. you see that the switch is not triggered. You even won't try to drive a car which has no tires ;)

closed 2022-07-24 by github-actions[bot]