knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #338: [FR] Alternative communication platfrom(s)

chilippso, 2022-06-05:
### Description

Unfortunately, it seems there is "just" this one FB group for asking questions and general support. I would really appreciate if there were (an)other platform(s) available, like an official IRC or Matrix channel or something.

### Feature Workflow

1. Think about suitable alternative communication platform(s)
2. Set up the new communication platform(s) 
3. Inform the users

#### Additional Information

I would recommend https://hackint.org as "service provider".

knutwurst, 2022-06-13:
Hi @chilippso ,
There is one (my) official Facebook group, that's right. But there are already more than 3300 people in here. There is also a discord server run by a friend. But this one is 100% in German and therefore maybe not ideal.

You have to keep in mind that I'm working on the project alone and the Facebook group is an ideal forum because you can help each other here. I can't concentrate on so many things at the same time since I also have a "real" job :)

Most questions can often be derived from the wiki or can be found directly in the official Marlin documentation.

An additional forum for which you have to register separately, I think is too much effort on my part. You are of course free to set up a forum yourself. :)

chilippso, 2022-06-13:
For me as a native german speaker, it's not the language that's problematic but the platform itself (Facebook) 😆

Is the Discord publicly available?

I would also agree that it would not make sense to setup a separate forum with registration, that's why I suggested something like an IRC channel, where people can just join - even just temporary with a Web-Based client (infrastructure is freely available like suggested in my initial post).

But as you said, there is already something the-like (the Discord) - even though one is required to have an account for that solution as well.

closed 2022-07-20 by github-actions[bot]