knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #341: [BUG] WLAN Anbindung funktioniert nicht (siehe #322)

RasputinBerry, 2022-06-13:
Hallo zusammen,
ich hatte die "MEGA_S_DGUS_BLT_10_v1.4.4" geladen um dann nach Nicht-Funktion der WLAN Anbindung den Beitrag #322 zu finden (In der Anleitung zum WLAN Einbau steht **AB**  und nicht **BIS** 1.3.0 ...  ist jetzt nicht wirklich ein BUG, aber ärgerlich (für Anfänger)).
Jetzt habe ich die "MEGA_S_DGUS_BLT_10_v1.3.0" aufgespielt und es geht leider noch immer nicht. 
Die Hardware des ESP-01 funktioniert am Programmer Stand-alone ohne Probleme, Verkabelung am Mainboard mehrfach kontrolliert. Ich finde keinen Fehler (mehr) ...

Muss man bei einem FW-Upgrade irgend etwas beachten ? Hat jemand eine Idee ?! Danke.

knutwurst, 2022-06-13:
Hallo @RasputinBerry ,
die Anleitung habe ich eben angepasst. Sorry dafür :/

Meistens liegt der Fehler an der falschen Baudrate oder daran, dass RX und TX vertauscht wurden.

RasputinBerry, 2022-06-13:
Hallo @knutwurst ,
das mit der Anleitung ist nicht (wirklich) schlimm, auf der Suche nach der Lösung lernt man immer wieder etwas dazu ... ;-) ... dauert halt nur ...

Zu Rx bzw. Tx mal eine **_blöde_** Frage: Auf deinen Fotos in der Anleitung ist Mainboard Rx mit ESP-01 Rx (bzw. dito. Tx) verbunden (s. Kabelfarbe). Sollte das gekreuzt sein ?!

knutwurst, 2022-06-17:
> Zu Rx bzw. Tx mal eine blöde Frage: Auf deinen Fotos in der Anleitung ist Mainboard Rx mit ESP-01 Rx (bzw. dito. Tx) verbunden (s. Kabelfarbe). Sollte das gekreuzt sein ?!

Nein, ich habe die Bezeichnung der Pins auf dem Board extra so gewählt. Die Bezeichnung heißt nicht "das hier ist TX", sondern "Hier muss TX angeschlossen werden".

Also wenn du die Farben verfolgst, sollte es genau so passen.

Du kannst natürlich mal versuchen, die Baudrate zu verringern. Dazu musst du dir allerdings die Firmware selbst kompilieren (hast du ja schon gemacht, wie du geschrieben hast). Versuch mal was ganz langsames. 9600 baud oder so.

RasputinBerry, 2022-06-21:
Aus purer Verzweiflung habe ich einiges "try & error" ausprobiert ... folgende (seltsame) Beobachtungen:
- Ich habe die Verkabelung verlängert um möglichst weit von irgendwelchen metallischen Konturen entfernt zu sein. 
Laptop bzw. SmartPhone sind innerhalb von einem Meter zum ESP.
- Der AP 192.168.0.1 ist im WLAN des SmartPhone, mit dem ESP am Drucker eingesteckt, (manchmal) erreichbar. Er reagiert aber nicht bzw. nur mit "Loading" oder "IP nicht erreichbar". Selten auch die WebUI, die aber immer einfriert bzw. nicht weiter ansprechbar ist (immer mit 1.3.0 )
- Im Laptop WLAN taucht bei derselben Konstellation kein ESP auf.
- Wie schon bekannt: Der ESP am Programmer und am USB Anschluss ist ohne Probleme im Heimnetz oder als AP zu finden.

Keiner eine Idee ?

closed 2022-07-27 by github-actions[bot]