knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #344: [BUG]Resume printing problems (short description)

XiaoYuYunshao, 2022-06-26:
你好 zusammen,
### Bug   After the machine stops after the material is cut off, click to resume printing, and printing cannot continue（MEGA_S_DGUS_TMC_v1.4.4.hex）

<Cut off stop    Power failure reply     Cannot continue printing>

### Steps to Reproduce

1. [First Step]  Click to pause printing during printing（or (it is detected that the consumables are gone when the material is cut off)）
2. [Second Step]  Click reply to print（The temperature is restored to the set value）
3. [and so on...] The accumulated time starts from zero and the machine does not continue to print

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]
    After printing for a few hours, the consumables were broken, the machine stopped, and the consumables were replaced, and the printing could not be continued, which was a little helpless. . .
#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

SargFowler, 2022-07-28:
I've got this issue too.
The printer pauses, ejects filament, I load new filament and then nothing.
I'm using PrusaSlicer which issues M600 to pause for filament change.
Are the special menu options required?

knutwurst, 2022-07-29:
Apologies for just now responding to this ticket. This is the same problem as described in ticket #350.

wk2uboot, 2022-07-30:
Hello,
First of all, thank you for the good work.
I own a 4Max Pro V2 and use the
4MAXP2_DGUS_TMC_BLT_10_v1.4.4 firmware.
Unfortunately, there is also the problem with the "Stop/Pause" function and resuming the job.
After pressing the "Pause" button, the head moves out of the print area to (-X, +Y) (left, back).
If you want to start the printer again, nothing happens and you have to start printing again.
I have not yet tested how it behaves after pressing the "filament switch" on the 4Max pro V2.
P.S.:
... I just tested the 4MAXP2_DGUS_TMC_10_v1.4.4 firmware (without "BLT"). The same problem exists here.

knutwurst, 2022-08-11:
@wk2uboot Since this is a hybrid firmware, which is built for all Anycubic printers that are based on an Atmega256, this (known) error unfortunately also exists with every printer :(

Unfortunately, I haven't been able to fix the error lately. However, in 1.4.4 it is possible to pause and resume printing via gcode/usb. Just not with the display.

wk2uboot, 2022-08-11:
Hello,
Thanks for the answer.
I hope that the problem will be solved for you later, because the printer should actually be a "stand-alone" solution.
Unfortunately I can't program otherwise I would have liked to help. I'm already glad that I managed to install the BL sensor.
Thanks again for your good work.

closed 2022-10-10 by github-actions[bot]