knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #346: [BUG] Autolevelling ignores Z endstops and breaks microswitch before failing

rgrgrg, 2022-07-10:
### Bug Description

When autolevelling with BLTouch there is no limit in Z movement below Z=0 - in case of problems with probe, head pushes down bed and both endstops. On my Mega X right endstop support bends because firmware tries to go too far down before "probe failed" message. 

It is related to faulty BLTouch connection but endstops should prevent mechanical damage in case of head going too far down.

### Steps to Reproduce

1. Remove  heatbed (or move it to Z=-20). 
2. Home (G28)
3. Autolevel (G29)
4. Gantry moves to mechanical lower limit. Right z endstop is displaced, sometimes rigth microswitch loses lever arm
5. "failed" message from firmware

**Expected behavior:** Fail about 2mm after triggering Z endstop (like in mesh levelling).

**Actual behavior:** Head moves until mechanical limit, bent right Z endstop support, damages switch (z_min)

#### Additional Information

Anycubic Mega X
Firmware 1.4.4

knutwurst, 2022-07-29:
I'm sorry. There was a bug, allowing the z probe to travel 12mm downwards instead of 2mm. It's fixed in the current build.

closed 2022-07-29 by knutwurst

borninstupidjuly, 2022-09-11:
Good afternoon.
For the first time, I am writing using GitHub.
Sorry if this is against the etiquette.
And I am translating from Japanese to English at DeepL, so I am sorry if the text is difficult to understand.

I would like to ask you a question about this matter.
I have installed 3dTouch on my MEGA-X and was setting it up when I came across this problem.

This issue is stated as fixed in the current build, but the one published in releases is 12 May 2022 and does not seem to be up to date.

For me, compiling the firmware is highly challenging.
If you have time, could you please upload a fix for this to the releases?
If this has already been fixed, I am very sorry.

Sorry for the length of this message. Thank you.

knutwurst, 2022-09-11:
Hi @borninstupidjuly ,
there is a new version on the way to release which coveres all the issues. :)

borninstupidjuly, 2022-09-11:
Hi @knutwurst 
Thank you for your prompt reply.
We look forward to your new release.

StefanB1970, 2022-10-28:
Hallo Knutwurst
Habe von [Printco](https://princore.de/collections/neu-druckplatten/products/flexplate-pei-rough-druckplatte-magnetsystem-pei-rough-federstahl-blech-inkl-aluplatte-silikonheizmatte?variant=43559915290888).
Mit der alten Firmware heitzt er das Bed in einen durch bis 100 Grad mit deirner  Firmware steigt er so bei 70 Grad aus. Durch reset und wieder aufheitzen komme ich irgendwann auf 100 Grad. Habe aber auch das Problem wenn ich den Druck Pausiere kommt er nicht zurück. Kannst du mir deine Firmware anpassen? Ich bekomme das nicht hin.

commit 626283aa (2022-07-29): Set Z_PROBE_LOW_POINT to -2 to fix #346