knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #348: [BUG] M412 S0 in startup script triggers runout warning

weirdgyn, 2022-07-26:
### Bug Description

On the print start the printer halts and warns about filament runout.

### Steps to Reproduce

1. Update the startup GCode of your silcer adding "M402 S0" command
2. Slice an object
3. Print the sliced object

**Expected behavior:**

The object is printed without any filament runout warning/stop.

**Actual behavior:**

Just after print start the printer halt and warn about a filament runout.
The warning is also triggered randomly along print.

#### Additional Information

Verified on Anycubic i3 Mega S with 1.0 board.
Knuthwurst release v1.3.1

knutwurst, 2022-08-11:
The internal filament runout feature does not work atm, because the filament sensor is handled by the TFT display and its internal logic. Also you are using an old firmware release :) 

If the filament runout is triggered often, you need to open the sensor and bend the little metal tingi a bit towards the filament. ...or disable it in the special menu (but this is only temporary and resets after power off)

FranciswritesCode, 2022-08-27:
Is this also the reason why it won't pause if the GCode is loaded using pronterface?

closed 2022-10-03 by github-actions[bot]