knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #349: [FR] Firmware with Wifi

elodur, 2022-07-26:
### Description

At last I managed to build the desired firmware with WiFi activated, but it took me quite some time to figure out I had to build a new firmware for the ESP-wifi to work.
Whatismore your picture of a 1MB ESP-01 was irritating since ESP3D does not fit in 1MB any more. Maybe you could add a hint to the already very good guide.

Thanks very much for your great guide und faq!

knutwurst, 2022-07-29:
Hi @elodur 
At the time I wrote the guide, there was no requirement for more memory. However, I have the current state + binary in my github, so it can be seen what I did.

Of course it's an older version, that's true. But this is what the guide refers to.
I have no influence on what the author of ESP3D does on his site.

But thanks for the hint! I'll keep it in mind :)

elodur, 2022-07-29:
This is no offense to you! I am very happy with your guide, just thought I could save someone else some time.

wk2uboot, 2022-08-25:
Hello Knutwurst,
Thank you for the instructions.
Perhaps one could also point out that the ESP01 USB programmer should be retrofitted with a button.
You need a button to set the programmer to programming status.

closed 2022-10-01 by github-actions[bot]