knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #350: Anycubic i3 Mega S - Filament-Sensor issue

harry7922, 2022-07-28:
Hi, I installed the latest firmware (knutwurst_MEGA_S_DGUS_v1.4.4.hex) on my i3 Mega S (with no modifications and DGUS Display like described in the HowTo).
After installing, I reseted to defaults and also removed filament and inserted it again.

When printing an object (with SD-Card), after round about 10 layers, the error message "lack of filament or filament sensor abnormal" occurs - but there is no issue with filament. Continuing print (by pressing "Continue" the print will not continue).
Additionally when stopping the print, in display the message "Stop success" popup over and over again (seems to be in infinite loop) also temperature of nozzle and heat bed is not lowered to "0".

As I understood, the filament sensor should be supported with this firmware - but seems to have trouble.
Hope I did not miss something in documentation.

Thanks in advance and regards,
Hardy

harry7922, 2022-07-28:
Just as an update:
I disabled the filament sensor in special menu and saved settings, but same issues. So actually I'm not able to print anything which needs more than round about 10 layers.

Thanks,
Hardy

knutwurst, 2022-07-29:
Hi @harry7922 ,
Unfortunately, this is a known problem. Resume (by pause or filament runout) can only be executed via gcode via USB. The display implementation is flawed at this point. But I'm working on it. The problem crept in with version 1.4.0. It still worked in 1.3.1.

By the way, you can only temporarily deactivate the filament sensor. So you have to do that before every print.

harry7922, 2022-07-29:
Hi @knutwurst,

thanks for the informations.
Then I know what I was doing wrong - I thought the deactivation of the filament sensor could be stored.
Thanks for your hint and for your work! Hope you will find a way to solve this problem.

Regards,
hardy

SargFowler, 2022-07-29:
What is the GCODE command to resume the print after a pause BTW?
I've tried M108, but this didn't work

Equilibri0, 2022-08-07:
Phasing the same Issue, thx for the workaround :)

PDiebolt, 2022-08-24:
Hello,

Thank you for your work. I am experiencing the same problem, with M600 it parks, retracts filament, prompts a change, after a change it feeds filament until the end, then does nothing. I downloaded [MEGA_S_DGUS_v1.4.4.hex]. I also found that after a filament shortage it does not continue to print. I understand you are working on it, thank you.

TechnicalAmazing, 2022-09-16:
Is the problem:”filament lack or missing” fixed?

slucero, 2022-10-05:
Hi, I'm facing this same issue as well where I cannot get the print to resume. I encountered this during a print from Cura where I added the script for a filament change using M600, but I've also been able to reproduce it directly with Pronterface outside the context of a running job. Here is the console output with debugging enabled in case it's helpful:

```
>>> m111 s247
SENDING:M111 S247
echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION
echo:M105
echo:M105
>>> M600
SENDING:M600
echo:M600
//action:paused
//action:prompt_end
//action:prompt_begin Pause
//action:prompt_button Dismiss
//action:prompt_show
echo:Insert filament and send M108
//action:prompt_end
//action:prompt_begin Nozzle Parked
//action:prompt_button Continue
//action:prompt_show
echo:busy: paused for user
echo:busy: paused for user
>>> M108
SENDING:M108
//action:prompt_end
//action:prompt_begin Paused
//action:prompt_button PurgeMore
//action:prompt_button Continue
//action:prompt_show
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
>>> M108
SENDING:M108
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
```

When I initially sent the `M600` command it started the filament change process by parking and then retracting the existing filament. Once I sent the first `M108` command it pulled in and readied the filament, but didn't move the print head at all. The final `M108` command received no response at all.

For reference, this is using a mostly stock Anycubic Mega Pro and the [MEGA_P_DGUS_v1.4.4.hex](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/download/1.4.4/MEGA_P_DGUS_v1.4.4.hex) firmware.

I believe it's already known as well, but the display on the printer never changed at any point to show a prompt button or message. I think I saw it mentioned this part was still a work in progress and that's why you had to use the gcode commands at the moment, but I wanted to add the observation just in case that interpretation was incorrect.

Thank you for your work on this, and if there's any more I can test to help get more info then I'm glad to help.

dwhittingham, 2022-10-13:
Adding some more info incase it is helpful.
I have a Mega Pro and used the beta firmware MEGA_P_DGUS_v1.5.0-b1.hex
I just ran a print and filament ran out. The print was not paused, just kept trying to print with no filament.

Extra notes for overall this is the first upgrade I have done on the printer. I am running it via USB from Octoprint running on a Ubuntu 20.04 VM running on a Proxmox hypervisor.

I have also noticed that the force sensitive resistor used for bed leveling no longer works. When trying to level the bed with it it used to give off a long beep whenever the FSR was being compressed by the nozzle and the bed. It now does not do that it gives a faint clicking sound and causes the head to retract like it crashed. It is then stuck until I power cycle.

And finally thanks so much for all your work on this.

harry7922, 2022-12-28:
Hi everyone,
any news about this issue?
Thanks!

Fantazzj, 2023-01-22:
Still today the error persist

Fantazzj, 2023-01-22:
> Hi everyone, any news about this issue? Thanks!

Still today the problem persist

RoyalCyan, 2023-02-16:
I have the same issue where it displays "lack of filament or filament sensor abnormal". However I can resume using the on-screen button. I must conduct more tests whether or not the button works all the time or partially.

SamuelRilling, 2023-05-06:
Is there still no fix for this? Pausing completely freezes the printer, sadly the beta version does not work either.
I saw a thread that fixed the pause.cpp file and some others, but it was specific for the Ender 3. Is there no such fix here?

Fantazzj, 2023-05-06:
At least is there a way to permanently disable the filament sensor without doing it every time at startup?

SamuelRilling, 2023-05-06:
> At least is there a way to permanently disable the filament sensor without doing it every time at startup?

You can simply change "#define FIL_RUNOUT_ENABLED_DEFAULT" to false on Configuration.h, but that of course requires you to compile the firmware yourself

knutwurst, 2023-05-24:
Fixed in #62fcc2f :) 
You can compile the current master, or wait for the final 1.5.0 release.

closed 2023-05-24 by knutwurst

SamuelRilling, 2023-05-25:
> Fixed in [62fcc2f](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/commit/62fcc2ff35c8cc17649e079fb2a2c70d77ae2b80) :) You can compile the current master, or wait for the final 1.5.0 release.

Finally fixed the M600 freeze bug compiling c01193b on the MEGA_S_TMC version. Thanks!