knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #351: [BUG] Filament Change while Printing by G-Code

Equilibri0, 2022-07-31:
Hey, 

first things first, great work :) i am so happy that i found your project, since yesterday i can print again because of the mesh leveling. 

Acutal problem or maybe my own stupidity: 

I have tried to print with two colors by using inside of Cura G-Code-> Filament Change Script on a given layer with my Mega Pro 
At least with your Custom for the first time simething happens, so it starts getting out the filament but then it stops at the middle between the heater and the engine which is moving the filament. I needed to get it manuel out with some force ( which i expect should not be the case ) , After geting the filament out i added the new one and its started to insert it BUT it stops right in front of the nozzle and the filling engine blocks and stopped working. 

Any suggestions ? is this a issue or did i used it just wrong ?

hopi11, 2022-08-09:
Hello @ all

I have the same Problem

Equilibri0, 2022-09-04:
Any kind of response woud be nice.

knutwurst, 2022-09-04:
Hi,
I'm sorry, but this is a known issue. See #344

Equilibri0, 2022-09-04:
> Hi,
> 
> I'm sorry, but this is a known issue. See #344 and #350

Thanks 😃 for the quick 
Response, alright.

closed 2022-10-10 by github-actions[bot]