knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #352: Is there a newer BLTouch wiring diagram for the Mega X V0.0.2  TriGorilla board?

Ri-Pete, 2022-08-02:
I bought a Mega X just recently and it has the TriGorilla board V0.0.2  without the logo and I was wondering if there are any differences I need to be aware of.
I purchased a BLTouch V3.1 and would like to install it. But before I do, I wanted to _**not**_ make any mistakes that I could avoid.
Also, does anyone know if the hot board is the same or has it also been upgraded?
[image]

knutwurst, 2022-08-09:
The Trigorilla 0.0.2 doesn't have a logo. Please see https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Beginner's-Guide-(English)#trigorilla-002 .

Ri-Pete, 2022-08-10:
OK, Thanks.

closed 2022-08-11 by knutwurst